AdnanHodzic/displaylink-debian #403: Installation fail

RicLzPt, 2020-04-30:
Hi dear!
I try install via method 1 on Ubuntu 20.04 (new install of Ubuntu)
But, occurr the error

```
  CC [M]  /home/riclzpt/tmp/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /home/riclzpt/tmp/displaylink-debian/evdi/module/evdi.ko
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
make[1]: Leaving directory '/home/riclzpt/tmp/displaylink-debian/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1]: Entering directory '/home/riclzpt/tmp/displaylink-debian/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
evdi_lib.c:6:10: fatal error: libdrm/drm.h: File or directory not exists
    6 | #include <libdrm/drm.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [<builtin>: evdi_lib.o] Error 1
make[1]: Leaving directory '/home/riclzpt/tmp/displaylink-debian/evdi/library'
make: *** [Makefile:9: all] Error 2
cp: não foi possível obter estado de '../library/libevdi.so.1.*': File or directory not exists
```
Sorry my porr english

mgcox2, 2020-05-01:
I'm having the same exact issue with Pop_OS! 20.04. Hopefully someone can chime in about this soon...

mgcox2, 2020-05-01:
@RicLzPt  Installing package libdrm-dev fixed the problem for me and I can now install successfully, thanks to the following comment: https://github.com/AdnanHodzic/displaylink-debian/issues/404#issuecomment-622342473

RicLzPt, 2020-05-01:
Thanks!!!!! Its works
[image]

closed 2020-05-01 by RicLzPt