AdnanHodzic/displaylink-debian #404: Pop_OS not supported

stuartmaxwell, 2020-05-01:
I'm guessing this should work since it's just an Ubuntu derivative. 

```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.0-7626-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed

---------------------------------------------------------------

Unsuported platform: Pop focal
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------
```

Erwin-D, 2020-05-01:
Try to insert the codeblock below right before https://github.com/AdnanHodzic/displaylink-debian/blob/31fa071f270cc0bd42d0a82c66096afc69f0f194/displaylink-debian.sh#L241

```
elif [ "$lsb" == "Pop" ];
then
	if [ $codename == "focal" ] || [ $codename == "n/a" ];
	then
		echo -e "\nPlatform requirements satisfied, proceeding ..."
	else
		message
		exit 1
	fi
```

stuartmaxwell, 2020-05-01:
Thanks @Erwin-D I actually did a crude and dirty hack on the script and got it to install successfully. Had to also install an extra package on Pop 20.04 to get it to work: libdrm-dev

Hope this feedback helps.

Erwin-D, 2020-05-01:
Thank you for the feedback. Would you mind testing the method I mentioned please?
Just run it as --debug to test if it would run correctly.

AdnanHodzic, 2020-05-01:
> Thanks @Erwin-D I actually did a crude and dirty hack on the script and got it to install successfully. Had to also install an extra package on Pop 20.04 to get it to work: libdrm-dev
> 
> Hope this feedback helps.

If it really helps, could someone please create a PR and simply add `libdrm-dev` to be [installed as part of dependencies](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L96)?

Thanks

RafiqC, 2020-05-03:
Thanks @Erwin-D, that addition in the script worked.

Also a big thanks to @AdnanHodzic, for sorting out DisplayLink support on Linux ;)

AdnanHodzic, 2020-05-03:
@RafiqC most welcome :)

Erwin-D, 2020-05-03:
Thank you for confirming. PR created.
I suppose this issue can be closed.

AdnanHodzic, 2020-05-03:
#411 merged with `master`, closing the issue.

closed 2020-05-03 by AdnanHodzic