AdnanHodzic/displaylink-debian #405: Fractional Scaling in Ubuntu 20.04

kwlockwo, 2020-05-01:
I realise this may not be actually an issue with `displaylink-debian` but when I turn fractional scaling on in the display settings and set it too 125% all my screens turn  mostly black with the occasional colour lines down them (like if you pick a resolution you screen can't handle).  I have to wait until the keep changes dialogue timeout and screen reset before I can do anything.

```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Iris Graphics 540 (rev 0a)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 #Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting
Provider 1: id: 0xa3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

I had to add the `DisplayLink` device and remove the `Intel` one to stop graphical screen glitches and [apply this advice](https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only) to stop the mouse cursor from flickering.

Other than the fractional scaling everything seems to be working well on Ubuntu 20.04.

MiChatz, 2020-05-02:
Hi!

It seems that we have a very similar setup.    
Although in my case I the second screen is not in being recognized.    
Do you have any suggestions? 

```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-28-generic

---------------- DisplayLink info ----------------

Driver version: 5.2.14
DisplayLink service status: up and running
EVDI service version: 1.6.4

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 540
VGA: Intel Corporation Iris Graphics 540 (rev 0a)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 
#Section "OutputClass"
    #Identifier  "DisplayLink"
    #MatchDriver "evdi"
    #Driver      "modesetting"
    #Option      "AccelMethod" "none"
#EndSection

 #Section "Device"
#    Identifier  "Intel"
#    Driver      "intel"
#EndSection

Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

```

Barabazs, 2020-07-19:
I'm afraid there's nothing we can do here. Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi).

closed 2020-07-19 by Barabazs