AdnanHodzic/displaylink-debian #408: displaylink-debian.sh doesn't run in Ubuntu 20.04 on latest commit (6aba905b74c3f718e9ce36467b1fc7f49ab98eb5)

Joeb3219, 2020-05-01:
After running `sudo ./displaylink-debian.sh` with the latest commit to `master` branch, I receive the error:

`./displaylink-debian.sh: line 99: syntax error near unexpected token `else'`

When checking out commit `31fa071f270cc0bd42d0a82c66096afc69f0f194`, however, things work fine, and installation works on Ubuntu 20.04.

clinte14, 2020-05-02:
This is a bug! Open displaylink-debian.sh and edit line 97-98:

elif [ "$lsb" == "PureOS" ];
        deps=(unzip linux-headers-$(uname -r) dkms lsb-release linux-source x11-xserver-utils wget libdrm-dev)

To look like this:
elif [ "$lsb" == "PureOS" ];
then
        deps=(unzip linux-headers-$(uname -r) dkms lsb-release linux-source x11-xserver-utils wget libdrm-dev)

Save and rerun:
sudo ./display-link.sh

Barabazs, 2020-07-19:
@Joeb3219 Are you still experiencing issues?

Barabazs, 2020-07-29:
Fixed in commit 2a2a37eb148e80ca1be5bdfbb2c1f7bbdfe95c9d

closed 2020-07-29 by Barabazs