AdnanHodzic/displaylink-debian #410: Installation fails: "could not create symlink /lib/modules/5.5.0-2-amd64/build/Kconfig"

loiselqw, 2020-05-03:
Dear community,

after updating to latest evdi https://github.com/DisplayLink/evdi.git and trying to reinstall this displaylink driver from this repo, I get following error during installation process:

`
Preparing for install

Archive:  DisplayLink_Ubuntu_5.2.zip
  inflating: 5.2/displaylink-driver-5.2.14.run  
  inflating: 5.2/DisplayLink USB Graphics Software for Ubuntu 5.2-Release Notes.txt  
  inflating: 5.2/LICENSE             
Creating directory displaylink-driver-5.2.14
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.2.14  100%  
ln: die symbolische Verknüpfung '/lib/modules/5.5.0-2-amd64/build/Kconfig' konnte nicht angelegt werden: Die Datei existiert bereits
`
Removing the Kconfig file and restarting the installation script has no positive effect.

Running debian testing (bullseye) with 5.5.0-2-amd64 kernel.

Thx a lot in adv for your support

fullettortuga, 2020-05-05:
I had the same issue. I am running Debian buster, not bullseye/testing.

What worked for me was to directly install the New Ubuntu DisplayLink 5.3 Official Driver, which came out yesterday May 4.

See [DisplayLink Official Drivers](https://www.displaylink.com/downloads/ubuntu) and [How to install](https://support.displaylink.com/knowledgebase/articles/684649).

Installed correctly but didn't get to work in Wayland, had to use xorg and the usual screen layout configuration: 
`xrandr --setprovideroutputsource 1 0 && xrandr --output DVI-1-0 --auto --right-of eDP1`

loiselqw, 2020-05-06:
ok after some struggling I was able to install the ubuntu one with following command:

`LC_ALL=C sudo ./displaylink-driver-5.3.0.28.run`

As described here: [https://www.displaylink.org/forum/showthread.php?s=b501467a2f07dbacb21b412b986a3b1b&p=89854#post89854](url)

but unfortunately no displays are recognized using xorg and lightdm

`xrandr --setprovideroutputsource 1 0 && xrandr --output DVI-I-1-1 --primary` leads in following output `warning: output DVI-I-1-1 not found; ignoring`

any ideas?

Thx in adv

fullettortuga, 2020-05-07:
can you report the output of `xrandr`?

loiselqw, 2020-05-07:
here we go:
`xrandr --setprovideroutputsource 1 0`

`xrandr --listproviders`

`Providers: number : 5
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 1 name:modesetting
Provider 1: id: 0x121 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xff cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xdd cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xbb cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting`

`echo "displays connected:" && xrandr | grep " connected" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`
and
`echo "displays active: " && xrandr | grep -E " connected (primary )?[1-9]+" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`

in both cases 

`displays active: 
eDP-1`

lsusb gives following output:

`Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0bda:0316 Realtek Semiconductor Corp. USB3.0-CRW
Bus 002 Device 005: ID 17e9:4307 DisplayLink USB3.0 Hub             
Bus 002 Device 002: ID 2109:8110 VIA Labs, Inc. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f2:b604 Chicony Electronics Co., Ltd Integrated Camera (1280x720@30)
Bus 001 Device 007: ID 413c:3012 Dell Computer Corp. Optical Wheel Mouse
Bus 001 Device 006: ID 03f0:0024 HP, Inc KU-0316 Keyboard
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 002: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

and /usr/share/X11/xorg.conf.d/20-displaylink.conf looks like this

`Section "Device"
  Identifier	"DisplayLink"
  Driver	"modesetting"
  Option	"PageFlip" "false"`

It seems to me the driver is installed correct, but somehow DVI-display via displaylink is not recognized ;-/(

with an older installation (displaylink-debian not the official Ubuntu-one) I was able to get it run with DVI-I-1-1 output.

But when trying to install the display-link debian script - I receive following output:

`--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.6.0-1-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-05-07 20:20:29--  https://www.displaylink.com/downloads/file?id=1576
Auflösen des Hostnamens www.displaylink.com (www.displaylink.com)… 54.77.143.153, 52.214.95.115, 34.252.102.94, ...
Verbindungsaufbau zu www.displaylink.com (www.displaylink.com)|54.77.143.153|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 303 See Other
Platz: http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1588876229&Signature=XSwCPeRFaipNK3lgbw%2Bor8xr3%2FI%3D [folgend]
--2020-05-07 20:20:29--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1588876229&Signature=XSwCPeRFaipNK3lgbw%2Bor8xr3%2FI%3D
Auflösen des Hostnamens assets.displaylink.com (assets.displaylink.com)… 52.218.28.187
Verbindungsaufbau zu assets.displaylink.com (assets.displaylink.com)|52.218.28.187|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 13889958 (13M) [application/octet-stream]
Wird in »DisplayLink_Ubuntu_5.3.1.zip« gespeichert.

DisplayLink_Ubuntu_5.3.1.zip      100%[===========================================================>]  13,25M  5,24MB/s    in 2,5s    

2020-05-07 20:20:32 (5,24 MB/s) - »DisplayLink_Ubuntu_5.3.1.zip« gespeichert [13889958/13889958]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  
ln: die symbolische Verknüpfung '/lib/modules/5.6.0-1-amd64/build/Kconfig' konnte nicht angelegt werden: Die Datei existiert bereits`

Thx for any recommondations

fullettortuga, 2020-05-07:
Looks installed correctly and recognizing the displaylink.

can you try 

1. `xrandr --setprovideroutputsource 2 0` and repeat step 3 (i.e. `echo "displays active: " && xrandr | grep -E " connected (primary )?[1-9]+" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`)
2. `xrandr --setprovideroutputsource 3 0` and repeat step 3 (i.e. `echo "displays active: " && xrandr | grep -E " connected (primary )?[1-9]+" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`)
3. `xrandr --setprovideroutputsource 4 0` and repeat step 3 (i.e. `echo "displays active: " && xrandr | grep -E " connected (primary )?[1-9]+" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`)

Does the other display appear?

loiselqw, 2020-05-07:
thanks a lot 

`xrandr --setprovideroutputsource 4 0 && echo "displays connected:" && xrandr | grep " connected" | sed -e "s/\([A-Z0-9]\+\) connected.*/\1/"`

made it.

it's running with following script as expected:
```

xrandr --setprovideroutputsource 4 0

/home/atiye/.screenlayout/off.sh
sleep 2
/home/atiye/.screenlayout/on.sh
#xrandr --output DVI-I-1-1 --primary 
#xrandr --output eDP-1 --off
```

closed 2020-05-07 by loiselqw

salamonn, 2020-05-08:
When installing New Ubuntu DisplayLink 5.3 Official Driver, I get

`dkms is already the newest version (2.6.1-4).
libdrm-dev is already the newest version (2.4.97-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Unsatisfied dependencies. Missing component: DKMS.
This is a fatal error, cannot install DisplayLink Linux Software.
`
I don't understand, since dkms is installed but then seems to be missing. I tried uninstalling and letting the script automatically fetch it, same result. Did you face the same problem?

follettortuga, 2020-05-09:
There's a few ways to automate that, depending on your setup.

- If you have a laptop where you use the display occasionally, I recommend the solution in the [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)
- If its a fixed display that you want to configure automatically when you log-in, for example you can add a file in /etc/X11/Xsession.d/ or in ~/.config/autostrart, but I recommend you to just Google `xrandr startup script`

salamonn, 2020-05-19:
> There's a few ways to _automate_ that, depending on your setup.
> 
>     * If you have a laptop where you use the display occasionally, I recommend the solution in the [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md)
> 
>     * If its a fixed display that you want to configure automatically when you log-in, for example you can add a file in /etc/X11/Xsession.d/ or in ~/.config/autostart, but I recommend you to just Google `xrandr startup script`

Thanks a lot - this post was a major help and I managed to set everything up. Just a question out of curiosity - is there a specific reason you highlighted the word "automate"? Would you not use it in this context?

follettortuga, 2020-05-19:
If you are plugging and unplugging the display (e.g. laptop connected to an external display), then every time you plug and unplug you need to run a little script to configure the setup as described in the [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md), its quick and easy, but it is not fully automated.

If your use case is a fixed display, then it is indeed automated.