AdnanHodzic/displaylink-debian #412: Clipping box around cursor

mnowotnik, 2020-05-03:
It looks like this issue is caused by this bug: https://bugs.launchpad.net/ubuntu/+source/plasma-workspace/+bug/1684240
It manifests itself when DisplayLink driver is used. 

Do you know any workaround?
What definitely works for me:

```
sudo service sddm restart
```

It's a bit annoying to log in twice, since this command logs out a user.

mnowotnik, 2020-05-03:
Additionally, the issue doesn't appear when the system is booted in "Recovery Mode" for some reason.

Barabazs, 2020-07-19:
Have you tried this [suggestion](https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only)?

Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi).

closed 2020-07-19 by Barabazs

mnowotnik, 2020-08-03:
I think I've installed official display link driver for Ubuntu 20 and the issue disappeared.