AdnanHodzic/displaylink-debian #413: Error! Invalid number of arguments passed.

xdontkillmeisuck, 2020-05-04:
looks like it compiles fine but the version number is wrong so it will not install.

5.4.0-7626-generic #30~1588169883~20.04~bbe668a~dev-Ubuntu SMP Wed Apr 29 21:31:55  x86_64 x86_64 x86_64 GNU/Linux

 CC [M]  /home/jeff/Downloads/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /home/jeff/Downloads/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/jeff/Downloads/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /home/jeff/Downloads/displaylink-debian/evdi/module/evdi.ko
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-7626-generic'
make[1]: Leaving directory '/home/jeff/Downloads/displaylink-debian/evdi/module'
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1]: Entering directory '/home/jeff/Downloads/displaylink-debian/evdi/library'
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.6.4 -lc -lgcc 
make[1]: Leaving directory '/home/jeff/Downloads/displaylink-debian/evdi/library'

Installing driver version: 5.3

DisplayLink Linux Software 5.3.0.28 install script called: install
Distribution discovered: Ubuntu 20.04 LTS

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
ERROR: Failed to install evdi/5.3.0.28 to the kernel tree.

RaiSaugat, 2020-05-04:
Faced the same issue.

dfluff, 2020-05-05:
Same issue, albeit with a slightly differently formatted error message:
`
Installing driver version: 5.3

DisplayLink Linux Software 5.3.0.28 install script called: install
Distribution discovered: Ubuntu 18.04.4 LTS

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
ERROR: Failed to install evdi/5.3.0.28 to the kernel tree.
`
The above shows Ubuntu 18.04.4, but I also saw the same with an Ubuntu 20.04 Live USB boot.

dfluff, 2020-05-05:
Same issue here, both under Ubuntu 18.04.4 and under a live Ubunto 20.04 boot.

I've re-pasted the error message but html-ised the less-than and greater-than symbols since they appear to have been swallowed when pasted.

`
DisplayLink Linux Software 5.3.0.28 install script called: install
Distribution discovered: Ubuntu 18.04.4 LTS

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Error! Invalid number of arguments passed.
Usage: add &lt;module&gt;/&lt;module-version&gt or
       add -m &lt;module&gt;/&lt;module-version&gt; or
       add -m &lt;module&gt; -v &lt;module-version&gt;
ERROR: Failed to install evdi/5.3.0.28 to the kernel tree.
`

dfluff, 2020-05-05:
I don't know enough to try and fix it, but the issue appears to be in displaylink-debian/5.3/displaylink-driver-5.3/displayling-installer.sh  with the line
`
dkms install "${EVDI}/module"
`

At the point this line is reached ${EVDI} is a temporary directory, but there is no file/directory called "module" under that directory, even though there were no errors preceding this point.

leyrer, 2020-05-07:
I created a pull request with code that fixed it for me on Mint 19.3.
https://github.com/AdnanHodzic/displaylink-debian/pull/417

leyrer, 2020-05-07:
Submitted a pull request which fixed this issue for me on "Linux Mint 19.3 Cinnamon". YMMV ;)
https://github.com/AdnanHodzic/displaylink-debian/pull/417

AdnanHodzic, 2020-05-08:
@dfluff changes from #417 have been merged with master, please pull the changes and try again.

Barabazs, 2020-07-19:
Issue appears to be resolved.
Please reopen this issue if it still persists.

closed 2020-07-19 by Barabazs

arpho, 2021-01-01:
I have the same issue on ubuntu 20.04

Barabazs, 2021-01-01:
@arpho Please open a new issue and provide a detailed description of your problem.