AdnanHodzic/displaylink-debian #414: Could not build on Ubuntu 20.04

jannik8x8, 2020-05-04:
Hi,

- Ubuuntu 20.04LTS
- Linux kernel: 5.4.0-28
- Docking Station: Icy Box IB-DK2241AC

I did a 
`git clone https://github.com/AdnanHodzic/displaylink-debian.git`
and tried 
`sudo ./displaylink-debian.sh`

I failed with

```
Unsatisfied dependencies. Missing component: libdrm.
This is a fatal error, cannot install DisplayLink Linux Software.

```
in install mode.

Checkiing dependencies was okay:

```
Checking dependencies

unzip is installed
linux-headers-5.4.0-28-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed

Platform requirements satisfied, proceeding ...

```

What is wrong?

robotard, 2020-05-05:
I'm seeing an issue too... I was actually working on some changes that were with version 5.2, but 5.3 also appears to have come from nowhere over night... Lets see if we can figure this out...

robotard, 2020-05-05:
evdi.sh should still be installing libdrm-dev uninterrupted:

```
#!/bin/bash

apt-get install -y libdrm-dev libelf-dev
```

Let me see what happens if I install them freshly as coincidentally have just reinstalled 20.04 today

luis3m, 2020-05-05:
I'm also experiencing a similar issue:

```
Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
ERROR: Failed to install evdi/5.3.0.28 to the kernel tree.
```

robotard, 2020-05-05:
Start of EVDI is :

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.zip
  inflating: 5.3/displaylink-driver-5.3.0.28.run  
Creating directory displaylink-driver-5.3.0.28
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.0.28  100%  

-------------------------------------------------------------------
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libdrm-dev is already the newest version (2.4.101-2).
libelf-dev is already the newest version (0.176-1.1build1).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

robotard, 2020-05-05:
Let me see if I can install EVDI as standalone...

robotard, 2020-05-05:
the evdi.sh is ultimately very simple... (I've added bits because the patching on 5.2 had already been updated by DisplayLink... ), but now they've just released 5.3... 

Can you do me a favour and download this to your displaylink-debian folder and :
cp evdi.sh evdi.sh.bak
mv evdi.txt evdi.sh

[evdi.txt](https://github.com/AdnanHodzic/displaylink-debian/files/4582099/evdi.txt)

robotard, 2020-05-05:
This just reverts some changes I just made, so we're all back to the same point to see if the issue was with the new 5.3 EVDI

robotard, 2020-05-05:
also before installing :
sudo rm -rf 5.3 resources/ DisplayLink_Ubuntu_5.3.zip displaylink.sh

After this... I'm seeing this to be a new bug introduced by a new EVDI...

luis3m, 2020-05-05:
I just did all that and I'm still experiencing the same issue :/

robotard, 2020-05-05:
It's a new build from DisplayLink (5.3) out on May 4th (Starwars day)

https://www.displaylink.com/downloads/ubuntu

robotard, 2020-05-05:
It's a new build from DisplayLink (5.3) out on May 4th (Starwars day)

https://www.displaylink.com/downloads/ubuntu

Seems to work straight out of the box...

luis3m, 2020-05-05:
Perfect, just tried out and it indeed work straight out of the box. Thank you for your guidance and help.

robotard, 2020-05-05:
I've got 3 displays straight up from the DisplayLink site's build on a fresh 20.04 install from today... Annoyingly just after fixing another issue right here... 

Seems like they've fixed some issues...

luis3m, 2020-05-05:
I'm using 4 displays and works perfectly!

jannik8x8, 2020-05-07:
> It's a new build from DisplayLink (5.3) out on May 4th (Starwars day)

> https://www.displaylink.com/downloads/ubuntu

> Seems to work straight out of the box...

I have installed the driver for 20.04 and it works. Thank You!
But I think nevertheless there are some problems: I am not able to do screen-mirroring. So I cannot display the same content on both screens. 
I use a Laptop and I have added a second screen via the Icy-Box-Docking-Station. When I close the Laptop the other screen becomes very slow and laggy.

I hope you can help me

robotard, 2020-05-08:
Probably a silly question, but you are using a USB 3.0 port right? The reason I ask is that I’m running an extended display across 3 screens from a dual output Orico adapter (about £23 from Ali Express - can offer you a nice signup bonus too 😉), but have also had 4 displays running as I’ve also got another single DisplayPort adapter...

That’s all from a USB 3.0 hub, and my laptop’s a pretty old Asus Zenbook UX31A from 2012 albeit with an i7...

I’ve not looked into mirroring, but have you got that working without the lid shut? Any particular reason you can’t just pull all your apps over to the other display and set your dock to that one? 

Sorry if a lame answer thus far, but I can’t live without a dual extended display, and my guess is if you’re finding yourself here then you’re likely at least a little techy?

As it’s a new full driver from DisplayLink themselves, have you tried their forums?

Do keep us posted. 

Cheers,

Rob

Barabazs, 2020-07-19:
> I have installed the driver for 20.04 and it works. Thank You!
> But I think nevertheless there are some problems: I am not able to do screen-mirroring. So I cannot display the same content on both screens.

Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi/)

> I use a Laptop and I have added a second screen via the Icy-Box-Docking-Station. When I close the Laptop the other screen becomes very slow and laggy.

Duplicate issue #393 There are some suggestions in that issue.

closed 2020-07-19 by Barabazs