AdnanHodzic/displaylink-debian #418: Failed installaion on Debian Buster

salamonn, 2020-05-08:
Hi Adnan and anyone who might be able to help,

I'm fairly new to both Debian and Git so please forgive any faux pas in reporting bugs (just let me know what information to provide).
I followed the installation and the post-installation guide, but it seems not to have worked, since I get

`root@DebianMu:/home/nela# ./displaylink-debian.sh --debug
-------------------------------------------------------------------
Starting Debug ...
Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
cat: /sys/devices/evdi/version: No such file or directory
`
and

`root@DebianMu:/home/nela# DisplayLink info
bash: DisplayLink: command not found`

I ran `uname -a` and this is the system information it delivers:
`root@DebianMu:/home/nela# uname -a
Linux DebianMu 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) x86_64 GNU/Linux
`

I'd appreciate any help at all (and thank you for the work you've invested in this so far).

salamonn, 2020-05-08:
The solution in https://github.com/AdnanHodzic/displaylink-debian/issues/410#issue-611357894 worked for me as well!

salamonn, 2020-05-08:
Although the solution in #410 had worked and the monitor worked perfectly, after restarting I get a blank monitor. Any advice whatsoever would be highly appreciated.

follettortuga, 2020-05-09:
See the [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md), hotplug doesn't work on Debian. So if you already managed to install the official DisplayLink driver following [#410 ](https://github.com/AdnanHodzic/displaylink-debian/issues/410), the issue is resolved.

Also see my response to your question there.

closed 2020-05-19 by salamonn