AdnanHodzic/displaylink-debian #419: Black screen when change resolution to 1366x768

qman-web-development, 2020-05-11:
my configuration:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1028:096a]
VGA: Intel Corporation UHD Graphics (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [1028:096a]
VGA: Intel Corporation UHD Graphics (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 0

-------------------------------------------------------------------

ScreamCZE, 2020-05-29:
Can you let me know what adapter/dock do you have? The latest DL-6950 chipset does not support this resolution, so this could be also a problem.

Barabazs, 2020-07-19:
Issue does not appear to be related to the use of this installer.

closed 2020-07-19 by Barabazs