AdnanHodzic/displaylink-debian #420: Ubuntu Focal Fossa Displaylink mouse and keyboard slow with primary display closed/turned off

ashishkurian, 2020-05-12:
Hello Folks,

I ran into a strange issue that after installing the DisplayLink latest drivers using the script, my mouse and keyboard is very slow if I close the my laptop display/turn of the built in display.

I upgraded to Focal from Ubuntu 18.04 LTS and it is not a fresh install. I previously had the Display Link installed and after the upgrade, I saw the my doc was not detected. That is when I installed it again using the instructions here.

`::>>sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x119 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------`
Previously when I had other troubles, I followed this workaround at : https://support.displaylink.com/knowledgebase/articles/1859356-x-server-crashes-on-nvidia-systems-when-displaylin

Thus I am having Xorg files at /usr/share/X11/xorg.conf.d

ashishkurian, 2020-05-12:
Just to add. There is actually no lag in mouse movement. Only the response of the mouse clicks are slow. Thus I think this is some issue related to Xserver

moritzknorr, 2020-05-16:
I am having similiar issues. [See here[](https://github.com/AdnanHodzic/displaylink-debian/issues/393)

Barabazs, 2020-07-19:
Possible duplicate of #393

closed 2020-07-19 by Barabazs