AdnanHodzic/displaylink-debian #421: Any plans for displaylink-arch?

Luc45, 2020-05-13:
Hey buddy,

Any plans for making a `displaylink-arch` version?

Thanks!

AdnanHodzic, 2020-05-17:
Hey :)

I was thinking about extending it to Red Hat based systems as well as Arch. But since I'm currently [looking for maintainers/developers for this project](https://github.com/AdnanHodzic/displaylink-debian#looking-for-maintainers-developers), I would really prefer if someone else did this. 

It really isn't a lot of work and is pretty straight forward, if you're up for it please give it a try and I'll be more then glad to help you/review code that goes in a PR.

Barabazs, 2020-08-05:
@Luc45 Can you post the output of: `sudo ./displaylink-debian.sh --debug`

closed 2020-10-11 by Barabazs