AdnanHodzic/displaylink-debian #422: Fail install Ubuntu 20.04

RicLzPt, 2020-05-14:
Guys, i'm here, again...
I Had installed this Displaylink git project and it worked fine with my AOC e1649fwu usb monitor.
I had to format my computer for other reasons and after new installation i tried to install again my monitor, doing the same actions that i did before. But... not working.
Dont there anything installed, only Ubuntu and DisplayLink. I run apt update and upgrade before.
The monitor work with garbage on it screen. (I will post a photo).
I don't found the Displaylink service after installation.
I have already uninstalled and installed again, to no avail.
Follow the debug  

`Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-29-generic

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
1.7.0
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 3rd
GF108M
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
NVIDIA Corporation GF108M [GeForce GT 635M] (rev a1)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Arquivo ou diretório inexistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x139 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x117 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
`

Some help?

RicLzPt, 2020-05-15:
I run a POP OS 20.04 fresh install.
The same error occurr. Do not install the Displaylink service.

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Pop
Release: focal
Kernel: 5.4.0-7626-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 3rd
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Arquivo ou diretório inexistente
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 5 name:modesetting
Provider 1: id: 0x143 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x121 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xff cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xdd cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xba cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

Barabazs, 2020-07-19:
@RicLzPt Have you tried creating the file manually?

closed 2020-08-05 by Barabazs