AdnanHodzic/displaylink-debian #423: dlm.service missing on Debian sid

pde, 2020-05-19:
Everything seems to work as expected on my system, except the dlm service is missing:

```--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 5.4.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Carbon
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.7-3

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x148 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x126 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x104 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xe2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

pde, 2020-05-19:
It looks as though neither `/etc/X11/xorg.conf.d/20-displaylink.conf` nor `/usr/share/X11/xorg.conf.d/20-displaylink.conf` is being set up by the script successfully, trying to figure out why...

pde, 2020-05-19:
So the "Carbon" subsystem, presumably associated with thinkpad carbon laptops, is possibly missing in [this line](https://github.com/AdnanHodzic/displaylink-debian/blob/master/displaylink-debian.sh#L719). Running `xorg_modesetting_newgen` manually doesn't let me see the `dlm` service (or see a monitor with `xrandr`), but I get this:

```
sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: sid
Kernel: 5.4.0-2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Carbon
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.7-3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x148 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x126 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x104 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xe2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
````

RicLzPt, 2020-05-20:
I had same problem in my notebook. The service do not was installed. Ubuntu and Pop OS
https://github.com/AdnanHodzic/displaylink-debian/issues/422

apdework, 2020-06-11:
@RicLzPt @pde did either of you manage to resolve this? I seem to be having the same issue.

RicLzPt, 2020-06-12:
> @RicLzPt @pde did either of you manage to resolve this? I seem to be having the same issue.

I couldn't solve it yet! Waiting

pde, 2020-07-17:
(I got further by hacking my way through the script, but never got that DisplayLink monitor running)

Barabazs, 2020-07-19:
@pde  Have you tried creating the file manually?

closed 2020-08-05 by Barabazs