AdnanHodzic/displaylink-debian #425: Weird Teleporting Mouse Cursor Issue

matthew-davis, 2020-05-21:
I've just upgraded to 20.04 and am experiencing the slowdown issue already mentioned. I am managing it with the suggested script [here](https://github.com/AdnanHodzic/displaylink-debian/issues/393#issuecomment-622949351)

So I have two external monitors that I access through a Dell D6000 dock and all appears to be well with the exception of about a 1" strip alongside the right-hand side on the first of the two monitors. In this zone a mouse cursor will kind of blink as I drag through this area, and will disappear completely if I stop moving the cursor. However, when it has disappeared from the first monitor it appears in the same corresponding place on the second monitor. 

Any ideas of things to try to help remove this feature would be appreciated. I've attached my debug screen below:

--------------- Linux system info ----------------
Distro: Ubuntu
Release: focal
Kernel: 5.4.0-31-generic

---------------- DisplayLink info ----------------
Driver version: 1.7.0
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------
Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:

-------------------- Monitors --------------------
Providers: number : 5
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x162 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x140 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x11e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xe9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
-------------------------------------------------------------------

matthew-davis, 2020-05-24:
So I've had a journey over the weekend, and this problem appears to rear it's head on Pop OS! 20.04 as well as when I returned to Ubuntu 19.10 so most likely related to the fact that I updated the DisplayLink driver. 

Being curious I attempted to rewrite a wee bit to see if I could load up the DisplayLink 5.2 driver (not certain,but I'm guessing that is what I was using before I updated) However it failed to build evdi:
ERROR (code 2): Unable to add evdi/5.2.14 to DKMS source tree

Any thoughts are always appreciated.

Barabazs, 2020-07-19:
Have you tried this [suggestion](https://support.displaylink.com/knowledgebase/articles/1843660-screen-freezes-after-opening-an-application-only)?

Your best bet is to file an issue [here](https://github.com/DisplayLink/evdi).

closed 2020-07-19 by Barabazs