AdnanHodzic/displaylink-debian #429: Installation does not create displaylink config or service, and screens do not connect

apdework, 2020-06-11:
Hi there

I have a new displaylink dock and am trying to connect with dual HDMI monitors to a Dell XPS 15 running Ubuntu 20.04. Having installed the drivers per the instructions and connected the monitors via the dock, neither the monitors nor laptop respond in any way. I have uninstalled and reinstalled the drivers many times, both from the official site and from here.

The only other indications that something may be wrong are that there does not appear to be a service running for the driver (most other posts I have seen reference `dlm.service` or `displaylink.service`), and no `20-displaylink.conf` has been created, which suggest to me that the installation was basically ineffectual. I have no idea how to resolve this, though.

Output of `sudo ./displaylink-debian.sh --debug`:

```--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-37-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 630
VGA: Intel Corporation UHD Graphics 630 (Mobile)
VGA (3D): NVIDIA Corporation TU117M [GeForce GTX 1650 Mobile / Max-Q] (rev a1)
X11 version: 1.20.8-2ubuntu2.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2c2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2a0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x27e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x25c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1df cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting```

IncPlusPlus, 2020-06-13:
I also noticed that the service and .conf file are missing, even after using the script to reinstall. However, my external display seems to be working anyways. I've also noticed a pattern where gnome crashes to the login screen which don't happen if my dock isn't plugged in but that's likely a separate issue.
```
--------------- Linux system info ----------------

Distro: Pop
Release: focal
Kernel: 5.4.0-7634-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Graphics
[GeForce
VGA: Intel Corporation UHD Graphics 630 (Mobile)
NVIDIA Corporation TU116M [GeForce GTX 1660 Ti Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x45 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0x2b1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x28f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x26d cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x24b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x229 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0
```

rrajkomar, 2020-07-02:
Same problem on Debian 10.
@IncPlusPlus You do not seem to be on the same distribution, maybe it is an issue related to Ubuntu / Debian

Barabazs, 2020-07-21:
@rrajkomar Have you looked at the part about mouse pointer? https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md Please open a separate issue with the necessary debug log.

Barabazs, 2020-07-21:
@apdework Did the suggestion by IncPlusPlus work for you?

rrajkomar, 2020-07-22:
@Barabazs this part (mouse flicker) does not relate to what I'm experiencing : A complete black screen with only the mouse pointer visible.
As for #293, with the current global health situation (covid) I still haven't been back to my office so I haven't been able to test it out yet

Barabazs, 2020-07-22:
> @Barabazs this part (mouse flicker) does not relate to what I'm experiencing : A complete black screen with only the mouse pointer visible.

>Should you experience problems with the display either remaining black, **only showing mouse pointer** or a frozen image of your main screen, then this could be due to Intel graphics driver interfering with displaylink.
>Reference: issue #68
```

rrajkomar, 2020-07-22:
>

I've tried all of the suggested approaches aside from the application of the evdi patch that needs to be done manually.
As the concerned machine is used for work, I cannot and will not try any manipulation of that kind since I don't even understand what it is doing.
I'd rather wait for a compatible version.
Thanks anyway.

rrajkomar, 2020-08-03:
@Barabazs fyi I just tried the evdi manipulation (https://github.com/AdnanHodzic/displaylink-debian/issues/293) and no changes.
I stilll have the same issue : pitch black screen on which only the mouse can be seen.
Any window that I try to drag to it seem to be there but hidden under the black screen.

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

rrajkomar, 2020-09-07:
Just uninstalled the version I had, pulled master and reinstall (after rebooting of course).
Same issue : black screen only mouse visible.

Note : the default installed 20-displaylink.conf did not work for me.
Because of my setup, I had to use my own files (see below) :

20-intel.conf
```
Section "Device"
    Identifier  "Intel"
    Driver "intel"
    Option "VSync" "false"
EndSection
```

30-displaylink.conf
```
Section "Device"
    Identifier  "DisplayLink"
    Driver "modesetting"
    Option "AccelMethod" "None"
    Option "PageFlip" "false"
EndSection
```

rrajkomar, 2020-09-30:
Just pulled last version, uninstalled, rebooted, installed, rebooted again and now my display link provider is no longer recognized
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: D

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.4.210
1.6.0+dfsg
4.4.24
5.1.26
DisplayLink service status: 
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/30-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```
[... 146 lines cut ...]
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_debug.o
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.ko
make[2] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-11-amd64 »
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/module »
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1] : on entre dans le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
WARNING: Version 1.6.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
```

Barabazs, 2020-10-11:
@rrajkomar Please repost the debug output.

rrajkomar, 2020-10-14:
Here you go : 
```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.4.210
1.6.0+dfsg
4.4.24
5.1.26
DisplayLink service status: 
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/30-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 2
Provider 0: id: 0x68 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 1 name:Intel
Provider 1: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

20-intel.conf
```
Section "Device"
    Identifier  "Intel"
    Driver "intel"
    Option "VSync" "false"
EndSection
```

30-displaylink.conf
```
Section "Device"
    Identifier  "DisplayLink"
    Driver "modesetting"
    Option "AccelMethod" "None"
    Option "PageFlip" "false"
EndSection
```

Barabazs, 2020-10-14:
You're still on v1.6.0 and displaylink-debian installs 1.7.0.
Did you uninstall everything properly?
Did you redownload displaylink-debian to reinstall?

rrajkomar, 2020-10-14:
I pulled 14 days ago and installed then.
Let me try and repull.
I'll keep you posted.

rrajkomar, 2020-10-14:
Ok so I juste uninstalled, reboot, pulled form the github repository and reinstalled. 
But it looks like it's still installing the 1.6.0 version :confused:

rrajkomar, 2020-10-14:
Here's the full output of the install : 
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: I

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2020-10-14 13:44:19--  https://www.displaylink.com/downloads/file?id=1576
Résolution de www.displaylink.com (www.displaylink.com)… 34.252.69.235, 52.214.244.113, 18.203.121.10, ...
Connexion à www.displaylink.com (www.displaylink.com)|34.252.69.235|:443… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1602676459&Signature=Wm5ro%2FqbHDpeNaUfFrJft1YLBwk%3D [suivant]
--2020-10-14 13:44:19--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1602676459&Signature=Wm5ro%2FqbHDpeNaUfFrJft1YLBwk%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.89.251
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.89.251|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 13889958 (13M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_5.3.1.zip »

DisplayLink_Ubuntu_5.3.1.zip                         100%[=====================================================================================================================>]  13,25M   141KB/s    ds 97s     

2020-10-14 13:45:57 (140 KB/s) — « DisplayLink_Ubuntu_5.3.1.zip » sauvegardé [13889958/13889958]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

[... 76 lines cut ...]
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_debug.o
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.ko
make[2] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-11-amd64 »
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/module »
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1] : on entre dans le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
Detected running Xorg session and connected docking station
Please disconnect the dock before continuing
Installation terminated.
```

Barabazs, 2020-10-16:
And the --debug output please.

rrajkomar, 2020-10-16:
Here's the debug output. I thought I had already sent it sorry.
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: d

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.4.210
1.6.0+dfsg
4.4.24
5.1.26
DisplayLink service status: 
EVDI service version: 1.6.0

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: Ltd.
Ltd.
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/10-nvidia.conf /etc/X11/xorg.conf.d/30-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Aucun fichier ou dossier de ce type
Contents:
 

-------------------- Monitors --------------------

Providers: number : 3
Provider 0: id: 0x24e cap: 0x1, Source Output crtcs: 4 outputs: 6 associated providers: 1 name:NVIDIA-0
Provider 1: id: 0xb4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 1 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

rrajkomar, 2020-10-16:
Note that the NVIDIA card is usually not there when I want to use the displaylink setup.
I'm not at the office atm so I'm not using the same config.

Barabazs, 2020-10-16:
> Note that the NVIDIA card is usually not there when I want to use the displaylink setup.
> I'm not at the office atm so I'm not using the same config.
What do you mean by that? Are you talking about 2 separate laptops?

rrajkomar, 2020-10-16:
No, no : same laptop, the NVIDIA card is inside the laptop docking station.
When on the dock I'm not using the displaylink device (not its configuration).

Barabazs, 2020-10-16:
@rrajkomar I have some free time right now to assist you. We can chat here https://gitter.im/displaylink-debian/community
It'll be faster than here.

Barabazs, 2020-10-16:
It looks like the uninstall does not remove all files and this probably messes up the reinstall procedure.
Try this:

1. `sudo ./debian-installer.sh --uninstall` 
2. Copy the output in a text document and name it uninstall.log
3. reboot
4. `modinfo evdi` (you should get ".... Module evdi not found")
5. locate evdi (copy the output for here)
6. locate displaylink (copy the output for here)

rrajkomar, 2020-10-18:
1. There is no debian-installer.sh file, I assumed you meant displaylink-debian.sh :
```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: Aucun fichier ou dossier de ce type

-------------------------------------------------------------------
```
4.
```
filename:       /lib/modules/4.19.0-11-amd64/updates/dkms/evdi.ko
license:        GPL
description:    Extensible Virtual Display Interface
author:         DisplayLink (UK) Ltd.
depends:        drm,drm_kms_helper
retpoline:      Y
name:           evdi
vermagic:       4.19.0-11-amd64 SMP mod_unload modversions 
parm:           initial_loglevel:Initial log level (int)
parm:           initial_device_count:Initial DRM device count (default: 0) (ushort)
parm:           enable_cursor_blending:Enables cursor compositing on user supplied framebuffer via EVDI_GRABPIX ioctl (default: true) (bool)
```

5.
```
/etc/modprobe.d/evdi.conf
/home/morningstar/Downloads/evdi-5.2.14-src.tar.gz
/lib/modules/4.19.0-11-amd64/updates/dkms/evdi.ko
/opt/web/GIT/displaylink-debian/evdi
/opt/web/GIT/displaylink-debian/evdi.sh
/opt/web/GIT/other/displaylink-debian/evdi.sh
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/evdi.tar.gz
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/x64-ubuntu-1604/libevdi.so
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/x86-ubuntu-1604/libevdi.so
/opt/web/GIT/other/displaylink-debian/resources/evdi-all-in-one-fixes.patch
/usr/lib/libevdi.so.0
/usr/lib/libevdi.so.0.1.6.0
/usr/share/doc/evdi-dkms
/usr/share/doc/libevdi0
/usr/share/doc/evdi-dkms/README.Debian
/usr/share/doc/evdi-dkms/changelog.Debian.gz
/usr/share/doc/evdi-dkms/copyright
/usr/share/doc/libevdi0/changelog.Debian.gz
/usr/share/doc/libevdi0/copyright
/usr/src/evdi-1.4.210
/usr/src/evdi-1.6.0+dfsg
/usr/src/evdi-4.4.24
/usr/src/evdi-5.1.26
/usr/src/evdi-1.6.0+dfsg/Kconfig
/usr/src/evdi-1.6.0+dfsg/Makefile
/usr/src/evdi-1.6.0+dfsg/dkms.conf
/usr/src/evdi-1.6.0+dfsg/evdi_connector.c
[... 65 lines cut ...]
/var/lib/dkms/evdi/5.1.26/4.19.0-9-amd64/x86_64/module/evdi.ko
/var/lib/dpkg/info/evdi-dkms.list
/var/lib/dpkg/info/evdi-dkms.md5sums
/var/lib/dpkg/info/evdi-dkms.postinst
/var/lib/dpkg/info/evdi-dkms.prerm
/var/lib/dpkg/info/libevdi0.list
/var/lib/dpkg/info/libevdi0.md5sums
/var/lib/dpkg/info/libevdi0.shlibs
/var/lib/dpkg/info/libevdi0.symbols
/var/lib/dpkg/info/libevdi0.triggers
```

6.
```
/etc/X11/xorg.conf.d/30-displaylink.conf
/etc/X11/xorg.conf.d/30-displaylink.conf.old
/home/morningstar/Bureau/displaylink-debian-install.log
/home/morningstar/Bureau/logs-displaylink
/home/morningstar/Bureau/graphics_cards_conf/30-displaylink.conf
/home/morningstar/Bureau/graphics_cards_conf/displaylink_confs
/home/morningstar/Bureau/graphics_cards_conf/displaylink_confs/10-displaylink.conf
/home/morningstar/Bureau/logs-displaylink/uninstall.log
/home/morningstar/Downloads/displaylink-driver-5.3.1.34.run
/opt/web/GIT/displaylink-debian + its contents (git clone)
```

Barabazs, 2020-10-18:
>     1. There is no debian-installer.sh file, I assumed you meant displaylink-debian.sh :

Yes, sorry about that.
 
> Uninstalling ...
> 
> bash: /opt/displaylink/displaylink-installer.sh: Aucun fichier ou dossier de ce type

Did you remove /opt/displaylink/ manually?

Please try running `sudo ./display-debian.sh --reinstall` and post the output.

rrajkomar, 2020-10-18:
Did you remove /opt/displaylink/ manually?
I don't think so. Can't remember.

Will run the install tomorrow morning at the office.

rrajkomar, 2020-10-19:
Morning,
Just ran the reinstall command. Same output (see below)
```
sudo ./displaylink-debian.sh --reinstall

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

bash: /opt/displaylink/displaylink-installer.sh: Aucun fichier ou dossier de ce type

-------------------------------------------------------------------

Performing clean-up
```

rrajkomar, 2020-10-19:
Shouldn't I run install first though ? (since I uninstalled yesterday)

Barabazs, 2020-10-19:
> Shouldn't I run install first though ? (since I uninstalled yesterday)

The uninstall failed because /opt/displaylink is missing.

rrajkomar, 2020-10-19:
rmmod: ERROR: Module evdi is in use

Barabazs, 2020-10-19:
`sudo modprobe -r evdi`

rrajkomar, 2020-10-19:
```
sudo dkms status evdi
evdi, 1.4.210: added
evdi, 1.6.0+dfsg, 4.19.0-6-amd64, x86_64: built
evdi, 4.4.24: added
evdi, 5.1.26, 4.19.0-10-amd64, x86_64: built
evdi, 5.1.26, 4.19.0-11-amd64, x86_64: installed
evdi, 5.1.26, 4.19.0-6-amd64, x86_64: built
evdi, 5.1.26, 4.19.0-8-amd64, x86_64: built
evdi, 5.1.26, 4.19.0-9-amd64, x86_64: built
```

Barabazs, 2020-10-19:
You'll have to remove them with:
```
sudo dkms remove evdi/1.4.210 --all
sudo dkms remove evdi/1.6.0 --all
```
and then provide output again for `sudo dkms status evdi`

rrajkomar, 2020-10-19:
Removed all version, output is now empty

Barabazs, 2020-10-19:
> Removed all version, output is now empty

Almost there :)

Now repost the output of 
```
locate -i evdi
locate -i displaylink
```

rrajkomar, 2020-10-19:
```
/etc/modprobe.d/evdi.conf
/home/morningstar/Downloads/evdi-5.2.14-src.tar.gz
/lib/modules/4.19.0-11-amd64/updates/dkms/evdi.ko
/opt/web/GIT/other/displaylink-debian/evdi.sh
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/evdi.tar.gz
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/x64-ubuntu-1604/libevdi.so
/opt/web/GIT/other/displaylink-debian/5.3.1/displaylink-driver-5.3.1/x86-ubuntu-1604/libevdi.so
/opt/web/GIT/other/displaylink-debian/resources/evdi-all-in-one-fixes.patch
/usr/lib/libevdi.so.0
/usr/lib/libevdi.so.0.1.6.0
/usr/share/doc/evdi-dkms
/usr/share/doc/libevdi0
/usr/share/doc/evdi-dkms/README.Debian
/usr/share/doc/evdi-dkms/changelog.Debian.gz
/usr/share/doc/evdi-dkms/copyright
/usr/share/doc/libevdi0/changelog.Debian.gz
/usr/share/doc/libevdi0/copyright
/usr/src/evdi-1.4.210
/usr/src/evdi-1.6.0+dfsg
/usr/src/evdi-4.4.24
/usr/src/evdi-5.1.26
/usr/src/evdi-1.6.0+dfsg/Kconfig
/usr/src/evdi-1.6.0+dfsg/Makefile
/usr/src/evdi-1.6.0+dfsg/dkms.conf
/usr/src/evdi-1.6.0+dfsg/evdi_connector.c
/usr/src/evdi-1.6.0+dfsg/evdi_cursor.c
/usr/src/evdi-1.6.0+dfsg/evdi_cursor.h
/usr/src/evdi-1.6.0+dfsg/evdi_debug.c
/usr/src/evdi-1.6.0+dfsg/evdi_debug.h
/usr/src/evdi-1.6.0+dfsg/evdi_drm.h
/usr/src/evdi-1.6.0+dfsg/evdi_drv.c
/usr/src/evdi-1.6.0+dfsg/evdi_drv.h
/usr/src/evdi-1.6.0+dfsg/evdi_encoder.c
/usr/src/evdi-1.6.0+dfsg/evdi_fb.c
/usr/src/evdi-1.6.0+dfsg/evdi_gem.c
/usr/src/evdi-1.6.0+dfsg/evdi_ioc32.c
/usr/src/evdi-1.6.0+dfsg/evdi_main.c
/usr/src/evdi-1.6.0+dfsg/evdi_modeset.c
/usr/src/evdi-1.6.0+dfsg/evdi_painter.c
/usr/src/evdi-1.6.0+dfsg/evdi_params.c
/usr/src/evdi-1.6.0+dfsg/evdi_params.h
/var/cache/apt/archives/evdi-dkms_1.6.0+dfsg-1_all.deb
/var/cache/apt/archives/libevdi0_1.6.0+dfsg-1_amd64.deb
/var/lib/dkms/evdi
/var/lib/dpkg/info/evdi-dkms.list
/var/lib/dpkg/info/evdi-dkms.md5sums
/var/lib/dpkg/info/evdi-dkms.postinst
/var/lib/dpkg/info/evdi-dkms.prerm
/var/lib/dpkg/info/libevdi0.list
/var/lib/dpkg/info/libevdi0.md5sums
/var/lib/dpkg/info/libevdi0.shlibs
/var/lib/dpkg/info/libevdi0.symbols
/var/lib/dpkg/info/libevdi0.triggers
```

locate -i displaylink
```
/etc/X11/xorg.conf.d/30-displaylink.conf
/etc/X11/xorg.conf.d/30-displaylink.conf.old
/home/morningstar/Bureau/displaylink-debian-install.log
/home/morningstar/Bureau/logs-displaylink
/home/morningstar/Bureau/graphics_cards_conf/30-displaylink.conf
/home/morningstar/Bureau/graphics_cards_conf/displaylink_confs
/home/morningstar/Bureau/graphics_cards_conf/displaylink_confs/10-displaylink.conf
/home/morningstar/Bureau/logs-displaylink/uninstall.log
/home/morningstar/Downloads/DisplayLink USB Graphics Software for Ubuntu 5.3.1.zip
/home/morningstar/Downloads/displaylink-driver-5.3.1.34.run
/opt/web/GIT/other/displaylink-debian
```

Barabazs, 2020-10-19:
Delete these files:
```
/etc/X11/xorg.conf.d/30-displaylink.conf
/etc/X11/xorg.conf.d/30-displaylink.conf.old
/opt/web/GIT/other/displaylink-debian
```

Now you can try to reinstall with the latest displaylink-debian.
```
git clone https://github.com/AdnanHodzic/displaylink-debian.git
cd displaylink-debian/ && sudo ./displaylink-debian.sh --install > new_install.log
```
And report back with the output of the install.

rrajkomar, 2020-10-19:
Here's the output log :
 ```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2020-10-19 14:17:07--  https://www.displaylink.com/downloads/file?id=1576
Résolution de www.displaylink.com (www.displaylink.com)… 18.203.121.10, 34.252.69.235, 52.214.244.113, ...
Connexion à www.displaylink.com (www.displaylink.com)|18.203.121.10|:443… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603110427&Signature=EWP%2BdlF6QhaF375LF3VRAJFws%2FA%3D [suivant]
--2020-10-19 14:17:07--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603110427&Signature=EWP%2BdlF6QhaF375LF3VRAJFws%2FA%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.56.155
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.56.155|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 13889958 (13M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_5.3.1.zip »

DisplayLink_Ubuntu_5.3.1.zip                         100%[=====================================================================================================================>]  13,25M   162KB/s    ds 85s     

2020-10-19 14:18:32 (160 KB/s) — « DisplayLink_Ubuntu_5.3.1.zip » sauvegardé [13889958/13889958]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

-------------------------------------------------------------------
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
libelf-dev est déjà la version la plus récente (0.176-1.1).
libdrm-dev est déjà la version la plus récente (2.4.97-1).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libjsoncpp1 libstartup-notification0 lightning
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 32 non mis à jour.
find: paths must precede expression: `displaylink-installer.patch'
find: possible unquoted pattern after predicate `-name'?
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
--2020-10-19 14:18:33--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Résolution de crazy.dev.frugalware.org (crazy.dev.frugalware.org)… 85.10.194.195
Connexion à crazy.dev.frugalware.org (crazy.dev.frugalware.org)|85.10.194.195|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 6947 (6,8K) [text/plain]
[... 65 lines cut ...]
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.ko
make[2] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-11-amd64 »
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/module »
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1] : on entre dans le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
Detected running Xorg session and connected docking station
Please disconnect the dock before continuing
Installation terminated.
```

Cannot reboot ATM though.

Barabazs, 2020-10-19:
You can't install displaylink while connected to the dock. You'll have to try the install again.

rrajkomar, 2020-10-19:
I unplugged the device and ran the installation again...

Still getting some worrysome output : 
```

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2020-10-19 15:25:14--  https://www.displaylink.com/downloads/file?id=1576
Résolution de www.displaylink.com (www.displaylink.com)… 34.250.82.133, 34.252.69.235, 52.214.244.113, ...
Connexion à www.displaylink.com (www.displaylink.com)|34.250.82.133|:443… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603114514&Signature=j36pEUv2d5n6m4Es5puAXn2DRlo%3D [suivant]
--2020-10-19 15:25:14--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603114514&Signature=j36pEUv2d5n6m4Es5puAXn2DRlo%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.21.146
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.21.146|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 13889958 (13M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_5.3.1.zip »

DisplayLink_Ubuntu_5.3.1.zip                         100%[=====================================================================================================================>]  13,25M   116KB/s    ds 1m 49s  

2020-10-19 15:27:04 (124 KB/s) — « DisplayLink_Ubuntu_5.3.1.zip » sauvegardé [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

-------------------------------------------------------------------
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
libelf-dev est déjà la version la plus récente (0.176-1.1).
libdrm-dev est déjà la version la plus récente (2.4.97-1).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libjsoncpp1 libstartup-notification0 lightning
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 32 non mis à jour.
find: paths must precede expression: `displaylink-installer.patch'
find: possible unquoted pattern after predicate `-name'?
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
[... 63 lines cut ...]
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_debug.o
  CC [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /opt/web/GIT/other/displaylink-debian/evdi/module/evdi.ko
make[2] : on quitte le répertoire « /usr/src/linux-headers-4.19.0-11-amd64 »
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/module »
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1] : on entre dans le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1] : on quitte le répertoire « /opt/web/GIT/other/displaylink-debian/evdi/library »
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux 10 (buster)
WARNING: Version 1.6.0 of EVDI kernel module is already running.
Please reboot before attempting to re-install DisplayLink Linux Software.
Installation terminated.
```

Barabazs, 2020-10-19:
The failed patches can be ignored. What bugs me is why and how evdi 1.6.0 got reinstalled (or wasn't uninstalled at all?).
Also, the install of display-link failed again...

rrajkomar, 2020-10-19:
Maybe I need to reboot for the unistallation to be complete ?

Barabazs, 2020-10-19:
Maybe.

You could also try to put the displaylink-installer.sh manually in /opt/... and then retry to uninstall with displaylink-debian.
<details>
<summary>Content of the file is </summary>

```
#!/bin/bash
# Copyright (c) 2015 - 2020 DisplayLink (UK) Ltd.

export LC_ALL=C
SELF=$0
COREDIR=/opt/displaylink
LOGSDIR=/var/log/displaylink
PRODUCT="DisplayLink Linux Software"
VERSION=5.3.1.34
ACTION=install
XORG_RUNNING=true

install_evdi()
{
  TARGZ="$1"
  MODVER="$2"
  ERRORS="$3"

  local EVDI
  EVDI=$(mktemp -d)
  if ! tar xf "$TARGZ" -C "$EVDI"; then
    echo "Unable to extract $TARGZ to $EVDI" > "$ERRORS"
    return 1
  fi

  echo "[[ Installing EVDI DKMS module ]]"
  (
    dkms install "${EVDI}/module"
    local retval=$?

    if [ $retval == 3 ]; then
      echo "EVDI DKMS module already installed."
    elif [ $retval != 0 ] ; then
      echo "Failed to install evdi/$MODVER to the kernel tree." > "$ERRORS"
      return 1
    fi
  ) || return 1
  echo "[[ Installing module configuration files ]]"
  cat > /etc/modules-load.d/evdi.conf <<EOF
evdi
EOF
  cat > /etc/modprobe.d/evdi.conf <<EOF
options evdi initial_device_count=4
EOF

  echo "[[ Installing EVDI library ]]"
  (
    cd "${EVDI}/library" || return 1

    if ! make; then
      echo "Failed to build evdi/$MODVER library." > "$ERRORS"
      return 1
    fi

    if ! cp -f libevdi.so "$COREDIR"; then
      echo "Failed to copy evdi/$MODVER library to $COREDIR." > "$ERRORS"
      return 1
    fi

    chmod 0755 "$COREDIR/libevdi.so"
  ) || return 1
}

[... 601 lines cut ...]
      ACTION="install"
      ;;

    uninstall)
      ACTION="uninstall"
      ;;
    *)
      usage
      ;;
  esac
  shift
done

if [ "$ACTION" == "install" ]; then
  install_dependencies
  check_requirements
  check_preconditions
  install
elif [ "$ACTION" == "uninstall" ]; then
  check_requirements
  uninstall
fi

```
</details>

rrajkomar, 2020-10-20:
Reboot was the trick. This morning I have the following output : 
```
-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-11-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: Y

Downloading DisplayLink Ubuntu driver:

--2020-10-20 09:32:10--  https://www.displaylink.com/downloads/file?id=1576
Résolution de www.displaylink.com (www.displaylink.com)… 18.203.121.10, 34.250.82.133, 34.252.69.235, ...
Connexion à www.displaylink.com (www.displaylink.com)|18.203.121.10|:443… connecté.
requête HTTP transmise, en attente de la réponse… 303 See Other
Emplacement : http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603179739&Signature=SZayxPowPq%2BAqnMu07Q33R1I3%2B8%3D [suivant]
--2020-10-20 09:32:19--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1603179739&Signature=SZayxPowPq%2BAqnMu07Q33R1I3%2B8%3D
Résolution de assets.displaylink.com (assets.displaylink.com)… 52.218.108.51
Connexion à assets.displaylink.com (assets.displaylink.com)|52.218.108.51|:80… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 13889958 (13M) [application/octet-stream]
Sauvegarde en : « DisplayLink_Ubuntu_5.3.1.zip »

DisplayLink_Ubuntu_5.3.1.zip                         100%[=====================================================================================================================>]  13,25M  5,82MB/s    ds 2,3s    

2020-10-20 09:32:21 (5,82 MB/s) — « DisplayLink_Ubuntu_5.3.1.zip » sauvegardé [13889958/13889958]

Removing prior: "5.3.1" directory

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  

-------------------------------------------------------------------
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
libelf-dev est déjà la version la plus récente (0.176-1.1).
libdrm-dev est déjà la version la plus récente (2.4.97-1).
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  libjsoncpp1 libstartup-notification0 lightning
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 40 non mis à jour.
find: paths must precede expression: `displaylink-installer.patch'
find: possible unquoted pattern after predicate `-name'?
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
--2020-10-20 09:32:23--  https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
Résolution de crazy.dev.frugalware.org (crazy.dev.frugalware.org)… 85.10.194.195
Connexion à crazy.dev.frugalware.org (crazy.dev.frugalware.org)|85.10.194.195|:443… connecté.
[... 128 lines cut ...]
-------------------------------------------------------------------

Performing post install steps

Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.3.1.zip" file
Removing redundant: "5.3.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
Reboot now? [Y/n] n

Reboot postponed, changes won't be applied until reboot

-------------------------------------------------------------------
```

Barabazs, 2020-10-20:
Does it work after rebooting?

rrajkomar, 2020-10-21:
Ok
So I botoed my system this working.
Sadly no it doesn't work at all : the 20-displaylink.conf file that was created (see below) doesn't seem to be recognized (xrandr --list-providers only show the intel line)

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

```
xrandr --listproviders
Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel
```

If I switch back to my previous file (which uses the following configuration), the provider is detected, the monitor can be started but still only displays a black screen with only the mouse pointer on top of it
(application windows seems to be there hidden within or under the black screen)
```
Section "Device"
    Identifier  "DisplayLink"
    Driver "modesetting"
    Option "AccelMethod" "None"
    Option "PageFlip" "false"
EndSection
```

Barabazs, 2020-10-21:
Can you repost the output of --debug?

rrajkomar, 2020-10-21:
Sure. Here you go :
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-12-amd64

---------------- DisplayLink info ----------------

Driver version: 1.4.210
1.6.0+dfsg
1.7.0
4.4.24
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Ltd.
VGA: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Note that the 20-intel.conf file contains the following : 
```
Section "Device"
    Identifier  "Intel"
    Driver "intel"
    Option "VSync" "false"
EndSection
```
And is required to make my setup with the nvidia card in the docking station at home work.
(Although the VSync option was only added to get to the point of the black screen)

Barabazs, 2020-10-21:
It may be wise to limit the configs while troubleshooting. Just append .bak to the intel  config.

Try to use the config below in /etc/X11/xorg.conf.d/20-displaylink.conf

```
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

```

rrajkomar, 2020-10-21:
Will try this at lunch. I can't do it ATM.

rrajkomar, 2020-10-21:
Thanks @Barabazs  you're the man !!!
After months of issues with this I can finally use my third screen again.
Yay !

rrajkomar, 2020-10-21:
For anyone facing a similar issu, my final display link config is this one : 
Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

All that left is try it with my intel config and see if it still works, if not I'll just adapt my script to handle the dock presence.

closed 2020-10-21 by Barabazs