AdnanHodzic/displaylink-debian #430: Monitor is not being detected post installation

engineer22, 2020-06-12:
Hello,

As a follow up of my previous [issue](https://github.com/AdnanHodzic/displaylink-debian/issues/428#issue-636555049) post-installation problems continue.

After rebooting the system I checked xrandr and the new monitor didn't apper. Running the debugger, I get these results after trying to apply [this method](https://github.com/AdnanHodzic/displaylink-debian/issues/236#issuecomment-471213411). I tried this method, and also [this one](https://github.com/AdnanHodzic/displaylink-debian/issues/228#issuecomment-467889348).

Running xrandr it doesn't find the new monitor, but running _xrandr --listproviders_ gives me 5 results. What should I try next? The debugging log is this (keep in mind that I alredy have an external monitor plugged in via HDMI, the problem occures on the new monitor which is an identicall one connected via HDMI-USB3):

-------------------------------------------------------------------
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Processor
VGA: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Graphics & Display (rev 0e)
VGA (3D):
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xe9 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0xc3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xa1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x7f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

pafribouf, 2020-06-14:
I think I have the exact same problem to the letter. I had to fix the ln problem you mentioned in the linked separate issue, xrandr doesn't show the device, the 20-displaylink.conf files don't exist, xrandr --listproviders lists 5 entries...

engineer22, 2020-06-14:
@pafribouf yes, I don't know.

I audit the code hopefully I can spot the problem and resolve it. I'll be posting the same issue on StackExchange too.

Barabazs, 2020-07-29:
@engineer22 Have you tried with this configuration?
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```

engineer22, 2020-07-29:
@Barabazs should I erase everything inside the 20-displaylink.conf files, and replace them with your suggestion?

Or I should add them in thr config?

Barabazs, 2020-07-29:
Yes, replace the content with what I suggested. But first, backup the files...

engineer22, 2020-07-29:
@Barabazs got it. I'll try it, and report back whether or not something changed.

engineer22, 2020-07-29:
@Barabazs UPDATE: After I made the changes, I'm unable to login.

I had to login via Wayland and revert the changes so I can login via xorg. I'm not sure if this is a good sign or not. Any guess on why it may be happening and any other suggestions?

Barabazs, 2020-07-29:
@engineer22 Is there anything in the file /etc/X11/xorg.conf ?

engineer22, 2020-07-29:
@Barabazs there is no such file.

I though I had to create it, tried to genetate it via init, and failed.

I then informed that you don't need that file at all, since we use files such as the 20-displaylink.conf, to create this type of configs.

Do you think I need it, and how do you generate it. I read a couple of tutorials, but init keeps saying that an xorg session is running.

TIA!

Barabazs, 2020-07-29:
> I then informed that you don't need that file at all, since we use files such as the 20-displaylink.conf, to create this type of configs.
> 
> Do you think I need it, and how do you generate it. I read a couple of tutorials, but init keeps saying that an xorg session is running.

Can't say for sure, but I don't think it's necessary. The reason I asked is, because according to previous issues, a difference between both files could lead to issues.

My last guess would be try with this config:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

engineer22, 2020-07-29:
@Barabazs I think that was my original config content, but I will try it again.

Did you see any other user using the config file you mentioned? I'm still not 100% sure if I need it or not, but I cant generate it.

engineer22, 2020-07-29:
@Barabazs just tried it. I login, but the resoult is same as the beginning of my issue.

Barabazs, 2020-07-29:
> @Barabazs I think that was my original config content, but I will try it again.
> 
> Did you see any other user using the config file you mentioned? I'm still not 100% sure if I need it or not, but I cant generate it.

https://github.com/AdnanHodzic/displaylink-debian/issues/200#issuecomment-453207457

> @Barabazs just tried it. I login, but the resoult is same as the beginning of my issue.

I'm out of ideas... Except maybe trying to uninstall, try to remove any leftover related files and retry the complete install procedure.

RationalG, 2020-08-01:
I am experiencing the same problem on Debian 10 (buster). Once the driver got installed, the service wasn't running and no config file was created. I ran the service & created the file manuallly but xrandr is totally unable to detect the monitor. 

Here is the --debug output : 

```
Distro: Debian
Release: buster
Kernel: 4.19.0-9-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
nvidia
Subsystem: 2nd
GeForce
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
NVIDIA Corporation GF116M [GeForce GT 555M/635M] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcb cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

```

At the bottom you can see by yourself that 5 providers are listed. Providers got associated to the output after I ran xrandr --setprovideroutputsource 1 0 (I did that for 2 0, 3 0, 4 0 aswell) but it didn't changed anything.

The displaylink device is listed in the `lsusb` output so I guess the device itself is not the problem...

rdmtinez, 2020-08-09:
I am having the same issue and I've tried everything at this point and nothing has worked. One thing I'm noticing right now is that unlike the --debug output in Adnan's example, is that in the above (and in mine) the DisplayLink Info only show's the EVDI version. Maybe its just a line of code that was forgotten and its no big deal. Second, in my case I noted that the DisplayLink driver that is installed is the version designed for Ubuntu 20.04 (5.3) while I'm currently on 18.04.5 and should be using 5.2. Again, I'm not sure if that is causing an issue. Lastly, I've  noted that using the 20-displaylink configuration files forces the provides to be shown even when (as it is in my case) the docking station or monitors are not plugged in; and xrandr --current 'recognizes' displays that aren't plugged in show up as disconnected

PS. @AdnanHodzic if you reading this there's a small bug in the script it occurs at the very end of the installation when the script asks whether it should restart now or later. If I say no, then and only then is the 20-displaylink.conf files written, but has the consequence of crashing my system and I have to REISUB,  and if I say 'yes' then the system restarts without writing the files. Note in my --debug output that there is not file created this was in the 'yes' case. 

`
------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.19.6-1ubuntu4.4

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcd cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 3 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

$ xrandr --current

Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
eDP-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 309mm x 173mm
   1920x1080     60.05*+  60.01    59.97    59.96    59.93  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)`

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

mrsloth000, 2020-09-05:
Hi, I´ve reinstalled with the latest version and this time the 20-displaylink.conf was created but unfortunately the screen is still not detected.
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
X11 version: 1.20.8-2ubuntu2.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "nvidia"
    Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
    Option "AccelMethod" "None"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2c2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2a0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x27e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x25c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1de cap: 0x2, Sink Output crtcs: 3 outputs: 3 associated providers: 1 name:modesetting

mrsloth000, 2020-09-05:
I'm a dummy - the screen was disconnected. Thank you for the fix!

closed 2020-09-11 by Barabazs