AdnanHodzic/displaylink-debian #431: Installation fails - Debian 10

randomRuser, 2020-06-13:
Hello,

I have been trying to install the displaylink drivers on my Debian 10 system in order to get my monitors running with a Lenovo docking station. I am quite new to Linux and cannot figure out what the problem is after putting in hours to solve the issue. 

The installation suddenly stops: 

`-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34  100%  
ln: failed to create symbolic link '/lib/modules/4.19.0-8-amd64/build/Kconfig': File exists
root@debian-nb-a:/opt/displaylink-debian# 
`

If I run the debugging, I get the following message: 

`-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory
`

Many thanks in advance!

engineer22, 2020-06-13:
Story of my life.

I had the same error and finally solved it myself after opening an issue and not getting any response.

Your error is this:

- ln: failed to create symbolic link '/lib/modules/4.19.0-8-amd64/build/Kconfig': File exists

What this means is that your **Kconfig** file cannot be created because is parent doesn't exist.
All you have to do is _cd /lib/modules/4.19.0-9-amd64/build_ and then make the parent: _sudo mkdir -p Kconfig_.

After that, the installation will successfully complete. Although your errors probably won't end there. If you see errors post-installation, follow my [issue#430](https://github.com/AdnanHodzic/displaylink-debian/issues/430#issue-637693957), and hopefully we'll get answers.

I'm audit the source code as we speak to resolve the issues I mention, and if I solve them I'll update.

AdnanHodzic, 2020-06-14:
@RameshPishini another bug report which has problem solution in and just needs to be implemented.

randomRuser, 2020-06-14:
Thank you egineer22! This solved the installation problem so that the DisplayLink drivers are running now. The display outputs of my Lenovo docking station are now detected by xrandr. 

However, still cannot use the displays via the dock and after putting in some more hours this afternoon, I have have to postpone finding a solution. I guess it is a Xfce issue. Any further advice is appreciated!

engineer22, 2020-06-14:
@randomRuser I'll be sure to update you if I find the solution.

randomRuser, 2020-06-25:
Hello, 

the drivers are working now. I think it was also an issue with Xfce. After uninstalling Xfce and installing KDE plus reinstalling the DisplayLink drivers and executing the following commands, I can finally use the monitors connected to the Lenovo dock with my Debian system: 

xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
xrandr --auto

Thank you very much for providing the Debian DisplayLink drivers! I donated some bucks to reward your awesome work.

closed 2020-06-25 by randomRuser