AdnanHodzic/displaylink-debian #433: Ubuntu 20 does not start after displaylink drivers install

vdhert, 2020-06-21:
Hi!

having all this:
```
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-37-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: 
Subsystem: GTX
VGA: NVIDIA Corporation TU117M [GeForce GTX 1650 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 4
Provider 0: id: 0x46a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0x448 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x426 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x404 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------
```

I have no troubles with installation at all, but after reboot machine hangs either showing splash screen or info that i Plymounth splash screen is OK and that's it.
I noticed that there are issues with UEFI, so i tried turning secure boot off, but it doesn't help (it should stay off though, right?).
I am able to run tty, uninstall drivers and after reboot everything is back to previous state, which is dock station works without extra screen.
I don't know much about linux, so excuse me if something here doesn't make sense.
Am i doing something wrong, or is this actual issue with this driver?

Barabas-dev, 2020-07-17:
You should include system logs around the time the issue occurs.

Barabazs, 2020-08-05:
Closing this issue due to inactivity.

Please reopen this issue if it still persists.

vdhert, 2020-08-05:
hey, I edited first post adding requested logs few days ago...

Barabazs, 2020-08-05:
An edit does not trigger a notification.
I'll have a look later today.

Barabazs, 2020-08-05:
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> Contents:
>  Section "OutputClass"
>     Identifier  "DisplayLink"
>     MatchDriver "evdi"
>     Driver      "modesetting"
>     Option      "AccelMethod" "none"
> EndSection

@vdhert  Was this created when installing or did you create that config manually?

> I noticed that there are issues with UEFI, so i tried turning secure boot off, but it doesn't help (it should stay off though, right?).

Yes, secure boot should be off (at least while installing and troubleshooting. Then you can try to sign the module yourself if you're feeling adventurous: [self-signing-modules.md](self-signing-modules.md)

vdhert, 2020-08-05:
this config was born automatically when I run install script.
I only made sure it was the only file in xorg.conf.d directory after first unsuccessful reboot (following advice from other thread in this repo), but that didn't help.

I keep secure not off when I occasionally try installing it. I think I will try that only after success with SB off, thank you though :)

Barabazs, 2020-08-06:
Can you reinstall with the latest version and report back with the output and if it works or not?

vdhert, 2020-08-09:
Hi @Barabazs 
sure!
what i have now:
```
-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: 
Subsystem: GTX
VGA: NVIDIA Corporation TU117M [GeForce GTX 1650 Mobile / Max-Q] (rev a1)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 4
Provider 0: id: 0x44e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 1: id: 0x42c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x40a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x3e8 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
-------------------------------------------------------------------
```
unfortunately it does not work. i made sure secure boot is disabled.
sys log attached.
[syslog09-08-2020.log](https://github.com/AdnanHodzic/displaylink-debian/files/5047321/syslog09-08-2020.log)

Barabazs, 2020-09-11:
@vdhert Please try again with the latest version. The latest commits might fix your issue.
Perform a proper uninstall with reboots at appropriate moment etc.

vdhert, 2020-09-21:
so i did, but it looks the same as before.
there seem to be new things in syslog though.
also: i don't know if that was the case before, but right now during install there are git warnings about being in detached head mode, here is sample:
[syslog21-09-2020.log](https://github.com/AdnanHodzic/displaylink-debian/files/5255053/syslog21-09-2020.log)

```Cloning into 'evdi'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Unpacking objects: 100% (46/46), 66.98 KiB | 1.05 MiB/s, done.
Note: switching to 'dc595db636845aef39490496bc075f6bf067106c'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
```

jdybdahl, 2020-10-11:
One of my computers refused to boot after I tried out various /etc/X11/xorg.conf.d/20-displaylink.conf files on Debian, so try first to comment out all lines and reboot to see if that helps.

Quite another thing: Ubuntu is actively supported by Displaylink.com and I have had Ubuntu 20.04 working on a computer. You should not need to use Adrian's otherwise excellent Debian script.

1. Get rid of every trace of displaylink on your computer: first uninstall then do a a sudo rm -r * (after checking if you have other stuff in that directory).

2. download displaylink from displaylink.com and that should just work.

JPGlaser, 2020-10-13:
@jdybdahl I am pleasantly surprised to hear that you have the Displaylink drivers working on 20.04 LTS. I haven't been able to get it to properly load, experiencing issues similar to @vdhert here where I can't get past the reboot for Xorg (nvidia-450 driver installed, which has been shown to work in the past with the latest 4.4 and 4.5 mainline kernels). 

How did you go about getting things working correctly? Do you have a descrete GPU on your system or an integrated one? Are you using the open-source or closed source nvidia drivers?

~ Joe G.

jdybdahl, 2020-10-13:
@JPGlaser, no, no GPU and no NVIDIA. I had Ubuntu on a Microsoft Surface Pro 3 laptop, upgraded it to 20.04  (not LTS) and installed DisplayLink from their site. The external screen switched on without me sending any commands. I have since installed Debian 10 on that computer and nothing worked until Barabazs fixed things.

jdybdahl, 2020-10-14:
@JPGlaser, have a look at https://github.com/AdnanHodzic/displaylink-debian/issues/473#issuecomment-707745318.

closed 2020-10-16 by Barabazs