AdnanHodzic/displaylink-debian #435: Please add support for Mint 20 Ulyana

ErnstAmat, 2020-06-29:


ErnstAmat, 2020-06-29:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-39-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: Bestand of map bestaat niet
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

Barabazs, 2020-07-19:
It looks like everything is installed as expected.

Please have a look at [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues)

ben-utzer, 2020-07-24:
It actually looks like this:

```Checking dependencies
unzip is installed
linux-headers-5.0.0-32-generic is installed
dkms is installed
lsb-release is installed
dpkg-query: package 'linux-source' is not installed and no information is avai
lable
Use dpkg --info (= dpkg-deb --info) to examine archive files.
linux-source not found! Install? [Y/n]
Reading package lists... Done
Building dependency tree
Reading state information... Done   
The following additional packages will be installed:
  linux-source-5.4.0
Suggested packages:
  libncurses-dev | ncurses-dev kernel-package libqt3-dev
The following NEW packages will be installed:
  linux-source linux-source-5.4.0   
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 134 MB of archives.
After this operation, 150 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-source-5
.4.0 all 5.4.0-42.46 [134 MB]
Get:2 http://archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-source a
ll 5.4.0.42.45 [2.440 B]
Fetched 134 MB in 14s (9.409 kB/s)  
Selecting previously unselected package linux-source-5.4.0.
(Reading database ... 283754 files and directories currently installed.)
Preparing to unpack .../linux-source-5.4.0_5.4.0-42.46_all.deb ...
Unpacking linux-source-5.4.0 (5.4.0-42.46) ...
Setting up linux-source-5.4.0 (5.4.0-42.46) ...
Setting up linux-source (5.4.0.42.45) ...
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

---------------------------------------------------------------

Unsuported platform: Linuxmint ulyana
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------```

ErnstAmat, 2020-07-26:
I did install the official version like ben-utzer stated. Sorry for my late reaction

closed 2020-07-29 by AdnanHodzic

ben-utzer, 2020-07-29:
Thanks, I will give it a try some time this week.

PR #446 (2020-07-29): Add support for Mint Ulyana
Resolve #435