AdnanHodzic/displaylink-debian #437: Monitor not detected on Ubuntu 20.04 fresh install

Cyb0rg42, 2020-07-01:
I have two monitor connected to my displaydock. The monitors are not detected by the displaylink driver. USB ports, Ethernet port are working fine.
I'm running a fresh install of Ubunt 20.04 on a Lenovo P52.

When running Windows (ugghh) the monitors are working.

Debug info:
--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-39-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: nvidia
Subsystem: P2000
VGA: NVIDIA Corporation GP107GLM [Quadro P2000 Mobile] (rev a1)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.1
X11 configs: /etc/X11/xorg.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x218 cap: 0x1, Source Output crtcs: 4 outputs: 6 associated providers: 4 name:NVIDIA-0
Provider 1: id: 0x2db cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2b9 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x297 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x275 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

Barabas-dev, 2020-07-17:
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
> Contents:
> 
You won't have a working setup without this file.

What is the content of  /etc/X11/xorg.conf ?

one-random-dev, 2020-09-03:
happened to me as well, fresh ubuntu 20.04 install, ran the script, installed successfully, rebooted, connected the USB, and I too am missing that file 
[image]

one-random-dev, 2020-09-03:
ok can add some additional information here

I am on an encrypted disk (not really sure if it affects the install process)
If I choose to reboot during installation, those files are not created 
I uninstalled rebooted and installed again selecting no on reboot bricked my laptop after the creation of some symlinks
forced shutdown (through the power button)
turned it on and connected the dock, no display recognized by linux yet
current debug output:
```
------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x47 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 5 associated providers: 5 name:modesetting
Provider 1: id: 0x160 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x13e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x11c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xfa cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0xd8 cap: 0x6, Sink Output, Source Offload crtcs: 5 outputs: 0 associated providers: 1 name:AMD Radeon Pro WX3100 @ pci:0000:3b:00.0

-------------------------------------------------------------------

```

Barabazs, 2020-09-12:
@one-random-dev Please try again with the latest version. The latest commits might fix your issue.
Perform a proper uninstall with reboots at appropriate moment etc.

one-random-dev, 2020-09-16:
really sorry but I cannot afford that atm
I really enjoy this tool and it saved me a lot of effort in the past 

However, for anyone using 20.04 Focal I would recommend downloading the updated version of the official driver, works like a charm 

Thanks

closed 2020-10-10 by Barabazs