AdnanHodzic/displaylink-debian #439: Clearly defining the scope of this tool

Barabazs, 2020-07-18:
Wouldn't it be useful to include a short, but clearly defined scope of this tool/repo?
ie. what type of issues can be reported, when can we classify the installation as successful, ...

Lots of issues don't seem to be related to this tool and should probably best be reported elsewhere.

AdnanHodzic, 2020-07-18:
I thought project README is self explanatory, same goes with [Troubleshooting most common issues](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#troubleshooting-most-common-issues) and [Post Installation Guide](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#post-installation-guide).

But even with that people will ask weird/unrelated questions. If you have a better idea please let me know, then if those bug reports are out of scope they can simply be closed.

Barabazs, 2020-07-21:
You're right, everything is mentioned in the those files. I guess some people just don't read it, same as with docs and manuals.

Barabazs, 2020-07-21:
Displaylink has dropped support for kernels < 4.15 https://github.com/DisplayLink/evdi/commit/d242444c656b5ba015267895e1ecc70fd1112371
Should issues concerning kernels < 4.15 be ruled as out of scope?

RavHilton, 2020-08-08:
My thinking is if DisplayLink has dropped support for kernels < 4.15, then absolutely issues concerning this is out of scope.

closed 2020-10-16 by Barabazs