AdnanHodzic/displaylink-debian #441: External display not working via dock station

jl2035, 2020-07-24:
I have Asus Laptop & Asus dock station.

I've installed the driver according to instructions.

I read the entire forum post here: https://www.displaylink.org/forum/showthread.php?t=67007

and installed the provided package from here:
https://www.displaylink.org/forum/showthread.php?t=67148

And I still can't get external display to work. I even tried all suggested fixes from the forum thread, uninstalled & installed again every time, but none of them worked :(

Can I please get some help?

DEBUG INFORMATION:

```
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: Picasso
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c4)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0
Provider 1: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xb3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x91 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Barabazs, 2020-07-26:
Please try to create `/etc/X11/xorg.conf.d/20-displaylink.conf` manually with this configuration:
```
Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection
```
Reboot, test and report back please.

jl2035, 2020-07-27:
Hi, it's still not working after restart. Same problem. So my external display connected directly to laptop is working, but the one connected to the dock station is not.

DEBUG INFORMATION: 

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: Picasso
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c4)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "AMDGPU"
    Driver     "amdgpu"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0

srkaysh, 2020-08-01:
Ensure the above mentioned conf file is present and then reboot. After your reboot, do the following steps:
```
modprobe udl

#set output source.. basically all the non-zero providers
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
...

#Go to your display settings and now see whether it detects or not.

```

jl2035, 2020-08-02:
I'm getting this:
```

j@laptop:~$ sudo modprobe udl
j@laptop:~$ xrandr --setprovideroutputsource 1 0
Could not find provider with index 1
j@laptop:~$ xrandr --setprovideroutputsource 2 0
Could not find provider with index 2
j@laptop:~$ 
```

jl2035, 2020-08-11:
This really seems like a serious problem far above my understanding.

There are new issues opened with the same problem: https://github.com/AdnanHodzic/displaylink-debian/issues/449 https://github.com/AdnanHodzic/displaylink-debian/issues/451

It's very sad. I invested some money to setup my workstation months ago, but now I have to deal with this :(
Is there any estimation, when can we expect the solution? Should I just sell my laptop and dock station and think of something else? Ubuntu OS is a must have for me :(

An ugly workaround would also be OK for me at this point.

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

jl2035, 2020-09-05:
@Barabazs 

I uninstalled, cloned the repo master branch and installed again. The monitor is still not showing anything.

j@laptop:~$ xrandr 
Screen 0: minimum 320 x 200, current 3600 x 1080, maximum 16384 x 16384
eDP connected 1920x1080+1680+0 (normal left inverted right x axis y axis) 344mm x 193mm
   1920x1080     60.00*+
   1680x1050     60.00  
   1280x1024     60.00  
   1440x900      60.00  
   1280x800      60.00  
   1280x720      60.00  
   1024x768      60.00  
   800x600       60.00  
   640x480       60.00  
HDMI-A-0 connected primary 1680x1050+0+0 (normal left inverted right x axis y axis) 474mm x 296mm
   1680x1050     59.88*+  59.95  
   1280x1024     75.02    60.02  
   1440x900      59.90  
   1280x960      60.00  
   1280x800      59.88  
   1152x864      75.00  
   1280x720      59.88  
   1024x768      75.03    60.00  
   832x624       74.55  
   800x600       75.00    60.32    56.25  
   640x480       75.00    59.94  
   720x400       70.08  
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)

Barabazs, 2020-09-05:
Post the debug output please.

jl2035, 2020-09-06:
@Barabazs here is the output:

> Starting Debug ...
> 
> Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
> 
> Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
> 
> --------------- Linux system info ----------------
> 
> Distro: Ubuntu
> Release: focal
> Kernel: 5.4.0-45-generic
> 
> ---------------- DisplayLink info ----------------
> 
> Driver version: 1.7.0
> 5.2.14
> DisplayLink service status: up and running
> EVDI service version: 1.7.0
> 
> ------------------ Graphics card -----------------
> 
> Vendor: amdgpu
> i2c_amd_mp2
> Subsystem: Picasso
> Inc.
> VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c4)
> VGA (3D): 
> X11 version: 1.20.8-2ubuntu2.3
> X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
> 
> -------------- DisplayLink xorg.conf -------------
> 
> File: /etc/X11/xorg.conf.d/20-displaylink.conf
> Contents:
>  Section "OutputClass"
>     Identifier  "DisplayLink"
>     MatchDriver "evdi"
>     Driver      "modesetting"
>     Option      "AccelMethod" "none"
> EndSection
> 
> -------------------- Monitors --------------------
> 
> Providers: number : 5
> Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0
> Provider 1: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> Provider 2: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> Provider 3: id: 0xb3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> Provider 4: id: 0x91 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
> 
> -------------------------------------------------------------------
> 
> Use this information when submitting an issue (http://bit.ly/2GLDlpY)

Barabazs, 2020-09-06:
Are you trying this with both a directly connected monitor **and** a monitor connected to your docking station?
If yes, then only try with the docking station for troubleshooting purposes.

jl2035, 2020-09-06:
Yes, that's correct.

I removed the other monitor now, so I'm only trying to use the dock station. I uninstalled, checked if the xorg file was gone and then installed again. Here is the debug information:

```

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-45-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.2.14
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: amdgpu
i2c_amd_mp2
Subsystem: Picasso
Inc.
VGA: Advanced Micro Devices, Inc. [AMD/ATI] Picasso (rev c4)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.3
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0
Provider 1: id: 0xe5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xc3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xa1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x7f cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
```

kngharv, 2020-10-10:
Take a look at this and see if it works for  you:
https://github.com/AdnanHodzic/displaylink-debian/issues/469

jdybdahl, 2020-10-11:
You shouldn't really need displaylink-debian to make Ubuntu work as Ubuntu (unlike Debian) is actively supported by Displaylink. I downloaded displaylink directly from displaylink's website to my Surface Pro 3 after having upgraded to the latest version of Ubuntu. It happily switched on my external monitor without any other effort from me.

However, I would first try to comment out each line in /etc/X11/xorg.conf.d/20-displaylink.conf, then disconnect your dock, reboot and connect the dock when rebooting has finished. 

If this doesn't work: google how to upgrade to Ubuntu's latest version (20.04) and then download displaylink from displaylink.com. It should just magically work. :)

If that doesn't work either: can you test the dock on another computer with Ubuntu or Windows?

Best of luck.

closed 2020-10-11 by Barabazs

jl2035, 2020-10-12:
Sadly there's only dark magic for me. I cannot get this to work.

@Barabazs do you have any idea, where can a person in my situation get some more help? Purchasing this dock really became a nightmare for me. It has already cost me hours and hours of no progress.

jdybdahl, 2020-10-12:
Can I assume that you erased ALL traces of displaylink-debian on your computer before trying the procedure here:
https://support.displaylink.com/knowledgebase/articles/684649

jl2035, 2020-10-18:
@jdybdahl sorry, I've missed your comment.

I assume I removed displaylink-debian correctly. Is there a way to make sure?

Output is now:

> xrandr --listproviders 
> Providers: number : 1
> Provider 0: id: 0x54 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 0 name:Unknown AMD Radeon GPU @ pci:0000:03:00.0

and

> xrandr 
> Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
> eDP connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 193mm
>    1920x1080     60.00*+
>    1680x1050     60.00  
>    1280x1024     60.00  
>    1440x900      60.00  
>    1280x800      60.00  
>    1280x720      60.00  
>    1024x768      60.00  
>    800x600       60.00  
>    640x480       60.00  
> HDMI-A-0 disconnected (normal left inverted right x axis y axis)

So external display seems to not be detected...

jdybdahl, 2020-10-19:
I think it is important to remove all scripts, programs and directories to ensure nothing old survives that can possible interfere with the software you are about to install.

I have the displaylink software in a separate directory: I uninstalled using the displaylink-debian script and then removed all files and directories with very dangerous 

> sudo rm -r * 

(assuming you have NOTHING you want to keep in that directory and subsirectories). Then reboot to remove evdi from memory.

jl2035, 2020-10-19:
> I have the displaylink software in a separate directory: I uninstalled using the displaylink-debian script and then removed all files and directories...

Where did you execute rm -r * ? I don't remember specifying the installation directory, so I guess I used the default.

jdybdahl, 2020-10-19:
Definitely only do it if you made your own subdirectory!!!

It is actually very few. These are the files you need to get rid of:

> evdi.sh
> displaylink.sh
> displaylink-debian.sh
> 

PLUS EVERYTHING IN THE resources DIRECTORY:

> evdi-all-in-one-fixes.patch
> eaiof.patch
> displaylink-installer.patch

jl2035, 2020-10-19:
Ok, so currently xrandr outputs this:

> Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 16384 x 16384
> eDP connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 344mm x 193mm
>    1920x1080     60.00*+
>    1680x1050     60.00  
>    1280x1024     60.00  
>    1440x900      60.00  
>    1280x800      60.00  
>    1280x720      60.00  
>    1024x768      60.00  
>    800x600       60.00  
>    640x480       60.00  
> HDMI-A-0 disconnected (normal left inverted right x axis y axis)
> DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
> DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
> DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
> DVI-I-1-1 disconnected (normal left inverted right x axis y axis)

Setprovideroutputsource still doesn't do anything:

> xrandr --setprovideroutputsource 1 0
> xrandr --setprovideroutputsource 2 0
> xrandr --setprovideroutputsource 3 0
> xrandr --setprovideroutputsource 4 0

jdybdahl, 2020-10-19:
Did you install DisplayLink from https://www.displaylink.com/downloads/ubuntu?

That should just work (even without the setprovideroutputsource commands; those are really only necessary on non-Windows and non-Ubuntu systems).
If you still have problems, your dock may not be working: test it on another computer system.

jl2035, 2020-10-21:
@jdybdahl Unbelievable! It's not working with a Windoz laptop either!

So I should just return the dock, right?

jdybdahl, 2020-10-21:
I am surprised that Ubuntu 20.04 fails.

Then your last hope is DisplayLink's support if a such exists.