AdnanHodzic/displaylink-debian #442: SideTrak monitor not detected on Kubuntu 18.04 with Plasma

LinuxRocks2000, 2020-07-25:
Not really sure whats happening here.

No /etc/X11/xorg.conf.d/20-displaylink.conf, and it doesn't even register that the screen is connected. 
Output of xrandr: 
`
Screen 0: minimum 8 x 8, current 1366 x 768, maximum 32767 x 32767
eDP1 connected primary 1366x768+0+0 (normal left inverted right x axis y axis) 340mm x 190mm
   1366x768      60.00*+  48.00  
   1360x768      59.96  
   1280x720      59.86    60.00    59.74  
   1024x768      60.00  
   1024x576      60.00    59.90    59.82  
   960x540       60.00    59.63    59.82  
   800x600       60.32    56.25  
   864x486       60.00    59.92    59.57  
   640x480       59.94  
   720x405       59.51    60.00    58.99  
   680x384       60.00  
   640x360       59.84    59.32    60.00  
HDMI1 disconnected (normal left inverted right x axis y axis)
VIRTUAL1 disconnected (normal left inverted right x axis y axis)`

Is it even possible for me to use this on Kubuntu 18.04? My kernel version is 5.something

The debug:
```
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Ubuntu
Release: bionic
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
5.1.26
5.2.14
DisplayLink service status: 
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 5500
VGA: Intel Corporation HD Graphics 5500 (rev 09)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2~18.04.1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------
```

Barabazs, 2020-07-26:
Please try to create `/etc/X11/xorg.conf.d/20-displaylink.conf` manually with this configuration:
```
Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection
```
Reboot, test and report back please.

LinuxRocks2000, 2020-07-26:
No change. I also rebooted with the SideTrak display plugged in the entire time, it showed its built-in boot screen and then went blank.

Tried xrandr and lsusb, no change in their output.

Barabazs, 2020-07-27:
So basically nothing happens when you plug it in and out?

Barabazs, 2020-08-05:
Issue is not related to displaylink-debian.

closed 2020-08-05 by Barabazs