AdnanHodzic/displaylink-debian #443: xorg_config_displaylink not created

Barabazs, 2020-07-26:
There appears to be a problem with the creation of "/etc/X11/xorg.conf.d/testfile.conf" as stated in multiple reports. I have personally experienced the same issue, and manually created the file with the required content. Then everything works as expected.

This issue should serve as a reference point and discussion to find the cause.

Barabazs, 2020-07-26:
Some of the reported issues are: #441 #442 #440 #437 #435 #427 #423

mrsloth000, 2020-08-05:
Thanks, I am also experiencing the same. Could you help me understand what us the required content to manually create the file?

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: [HD
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
X11 version: 1.20.8-2ubuntu2.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x1b8 cap: 0x1, Source Output crtcs: 0 outputs: 0 associated providers: 5 name:NVIDIA-0
Provider 1: id: 0x2c2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x2a0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x27e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x25c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1de cap: 0x6, Sink Output, Source Offload crtcs: 3 outputs: 3 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

RavHilton, 2020-08-08:
I have the exact same issue with Linuxmint ulyana - I have a Dell 7080 Micro system with a Dell D3100 DisplayLink dock.  I previously had an Intel Skull Canyon NUC and the same D3100 DisplayLink dock and was running Linux Mint 19.3 with the previous release of this DisplayLink driver and it was working fine.

I have tried both of the fixes listed in the Post Instalationl Guide regarding Intel iGPUs, neither of which resulted in the displays attached to the D3100 lighting up.  (Out of interest, I dual boot Win 10 on this same system and the displays work on that - so I know the D3100 is alive and kicking).

Starting Debug ...
Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------
Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------
Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------
Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9bc5 (rev 05)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2
/etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x108 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xc4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xa2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

kennethtran93, 2020-08-25:
If I recall correctly, another issue had said that for some reason if you accepted the initial restart by the displaylink run and not by the script itself, the X11 conf files are not written.  Try saying N to the first reboot, then Y to the second reboot.

canicola, 2020-08-25:
thanks for the comment. When should I be asked about the restart? I've tried to reinstall displaylink but was never asked

kennethtran93, 2020-08-26:
Here's my full installation log from the script.

While the script did run fine, I'm having a hard time getting my USBC ASUS portable monitor to work yet, but that's probably a displaylink issue and not this installation script.
```
--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.0-42-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Please read the Software License Agreement available at: 
https://www.displaylink.com/downloads/file?id=1576
Do you accept?: [Y/n]: y

Downloading DisplayLink Ubuntu driver:

--2020-08-25 18:46:10--  https://www.displaylink.com/downloads/file?id=1576
Resolving www.displaylink.com (www.displaylink.com)... 52.214.105.175, 52.214.95.115, 34.252.102.94, ...
Connecting to www.displaylink.com (www.displaylink.com)|52.214.105.175|:443... connected.
HTTP request sent, awaiting response... 303 See Other
Location: http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1598406971&Signature=E%2FhSxflaYaq9mfvc6nrpFaUxd7U%3D [following]
--2020-08-25 18:46:11--  http://assets.displaylink.com/live/downloads/software/f1576_DisplayLink%20USB%20Graphics%20Software%20for%20Ubuntu%205.3.1.zip?AWSAccessKeyId=AKIAJHGQWPVXWHEDJUEA&Expires=1598406971&Signature=E%2FhSxflaYaq9mfvc6nrpFaUxd7U%3D
Resolving assets.displaylink.com (assets.displaylink.com)... 52.218.88.131
Connecting to assets.displaylink.com (assets.displaylink.com)|52.218.88.131|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13889958 (13M) [application/octet-stream]
Saving to: ‘DisplayLink_Ubuntu_5.3.1.zip’

DisplayLink_Ubuntu_5.3.1.zip   100%[=================================================>]  13.25M  4.35MB/s    in 3.0s    

2020-08-25 18:46:15 (4.35 MB/s) - ‘DisplayLink_Ubuntu_5.3.1.zip’ saved [13889958/13889958]

-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Creating directory displaylink-driver-5.3.1.34
Verifying archive integrity...  100%   All good.
[... 130 lines cut ...]
Enable and start displaylink-driver service
Created symlink /etc/systemd/system/multi-user.target.wants/displaylink-driver.service → /lib/systemd/system/displaylink-driver.service.
Setup DisplayLink xorg.conf depending on graphics card
Wrote X11 changes to: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------------------------------------------------------------

Performing clean-up
Removing redundant: "DisplayLink_Ubuntu_5.3.1.zip" file
Removing redundant: "5.3.1" directory

-------------------------------------------------------------------

Install complete, please reboot to apply the changes
After reboot, make sure to consult post-install guide! http://bit.ly/2TbZleK
```
This is the second reboot prompt, this time generated by displaylink-debian script.
> Reboot now? [Y/n] *n*

This can be `Y` or `N`.  Just make sure all your other work has been saved/exited.
```
Reboot postponed, changes won't be applied until reboot

-------------------------------------------------------------------
```

closed 2020-09-05 by AdnanHodzic