AdnanHodzic/displaylink-debian #447: Unable to connect to a monitor in Kali Linux (via docking station)

srkaysh, 2020-08-01:
I've recently bought a docking station ( DELL D6000 452-BCYT ). It works perfectly with Windows OS but unable to detect the monitor in Kali Linux OS. The other components of the dock work fine.

As requested, below is the debug message from the script provided:
```
--------------- Linux system info ----------------

Distro: Kali
Release: kali-rolling
Kernel: 5.7.0-kali1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.0.0
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.8-2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x13b cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x119 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xf7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xd5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
```
-------------------------------------------------------------------

I've followed the post-installation steps and skimmed through the troubleshooting page too. I've also tried setting up the `/etc/X11/xorg.conf.d/20-displaylink.conf` file with different configs mentioned in the FAQs, still no luck.

Barabazs, 2020-08-01:
Have you tried this config?

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

srkaysh, 2020-08-01:
> Have you tried this config?
> 
> ```
> Section "OutputClass"
>     Identifier  "DisplayLink"
>     MatchDriver "evdi"
>     Driver      "modesetting"
>     Option      "AccelMethod" "none"
> EndSection
> ```

Yes, there was another config too. I've tried both.

srkaysh, 2020-08-01:
Ah, it works now. I'd to set the output source through xrandr.
```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0
```
My bad. Do I need to keep this issue open? since the xorg config file doesn't get generated through the script..

Barabazs, 2020-08-01:
Thanks for the feedback. I have linked this issue to another one that serves as the main issue.

closed 2020-08-01 by Barabazs