AdnanHodzic/displaylink-debian #448: Unsupported platform: Linuxmint ulyana

RavHilton, 2020-08-03:
Tried downloading the .zip and also cloning, but on a Linux Mint 20 install, this comes up with an error:
Unsupported platform: Linuxmint ulyana

even though the doc specifically states that this OS variant is supported.

Checking dependencies

unzip is installed
linux-headers-5.4.0-42-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

---------------------------------------------------------------

Unsuported platform: Linuxmint ulyana
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

RavHilton, 2020-08-03:
http://bit.ly/2GLDlpY

RavHilton, 2020-08-03:
-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9bc5 (rev 05)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x108 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xc4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xa2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

bortedev, 2020-08-03:
I can conform, got the same problem after the mint upgrade to 20 (ulyana).

mrsloth000, 2020-08-03:
I am here to report the same issue...

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: i

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-5.4.0-42-generic is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

---------------------------------------------------------------

**Unsuported platform: Linuxmint ulyana**
Full list of all supported platforms: http://bit.ly/2zrwz2u

This tool is Open Source and feel free to extend it
GitHub repo: https://github.com/AdnanHodzic/displaylink-debian/

---------------------------------------------------------------

~/displaylink-debian$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory

Barabazs, 2020-08-04:
What is the output of this:
`lsb_release -is`
`lsb_release -cs`

mrsloth000, 2020-08-04:
:~$  lsb_release -is
Linuxmint
:~$  lsb_release -cs
ulyana

RavHilton, 2020-08-05:
lsb_release -is and lsb_release -cs give the exact same results as shown in the Debug output:

--------------- Linux system info ----------------
Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

Barabazs, 2020-08-05:
Please replace your current `displaylink-debian.sh` file with [this](https://raw.githubusercontent.com/Barabazs/displaylink-debian-1/master/displaylink-debian.sh) one and try again.

mrsloth000, 2020-08-05:
Thanks, that resolved the installation issue but unfortunately the monitor isn't detected. I will troubleshoot later today.

closed 2020-08-05 by AdnanHodzic

RavHilton, 2020-08-07:
OK, it now installs (thanks for that), but there are no monitors detected on the DisplayLink device - seems to be a similar issue to that which mrsloth000 is also having.

RavHilton, 2020-08-07:
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 
VGA: Intel Corporation Device 9bc5 (rev 05)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x48 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 6 associated providers: 4 name:modesetting
Provider 1: id: 0x108 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xe6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xc4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xa2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

PR #450 (2020-08-05): Correct typo in Mint platform
Resolves #448