AdnanHodzic/displaylink-debian #449: Cannot connet to external monitor via Lenovo Dock

geblukasz, 2020-08-05:
Hi, 

I have run all commands from tutorial and still cannot connect to external monitor. Please provide with some solutions

--------------- Linux system info ----------------

Distro: Ubuntu
Release: focal
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x46 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 4 associated providers: 4 name:modesetting
Provider 1: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0xf6 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xd4 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

garrit-schroeder, 2020-08-05:
Having the same exact error. 

The following directory is missing on my system:
/etc/X11/xorg.conf.d

Barabazs, 2020-08-05:
@geblukasz
Create the config manually with this content:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

closed 2020-10-10 by Barabazs