AdnanHodzic/displaylink-debian #451: Monitors not detected on ICY BOX Notebook DockingStation

AlainKnaff, 2020-08-07:
Hi, I've got 2 monitors connected to my ICY BOX Notebook DockingStation.

Via DVI:  ColorEdge CG21
Via HDML: Terra 2463W

None of both show up spontaneously.
After xrandr --setprovideroutputsource $provider 0 on all 4 providers (1-4), the outputs do show up, but disconnected.
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)
DVI-I-3-3 disconnected (normal left inverted right x axis y axis)
DVI-I-2-2 disconnected (normal left inverted right x axis y axis)
DVI-I-1-1 disconnected (normal left inverted right x axis y axis)

alain@alainpc:~/tmp/displaylink-debian$ sudo ./displaylink-debian.sh --debug
[sudo] password for alain: 

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "InteIntel"
    Driver     "i915"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

alain@alainpc:~/tmp/displaylink-debian$ sudo ./displaylink-debian.sh --debug

-------------------------------------------------------------------

Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation UHD Graphics 620 (Whiskey Lake) (rev 02)
VGA (3D): 
X11 version: 1.20.4-1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier "Intel"
    Driver     "i915"
    Option     "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------

Use this information when submitting an issue (http://bit.ly/2GLDlpY)

-------------------------------------------------------------------

N.B. I had to create the /etc/X11/xorg.conf.d/20-displaylink.conf file manually. It didn't get created automatically, nor does the directory /etc/X11/xorg.conf.d/ exist by default on Debian 10.

Installation in general was a pain. This is for a work machine used at a home office, so it's configured by default for VPN without direct access to the internet. No direct internet is not an issue for most other programs, as they are smart enough to use a proxy (socks, http) where available. But not display-link debian. And to make matters worse, it tends to want to use the network all the time, for no good reason. For instance, why on earth is network access needed for --debug?

Then, failures are hard without any possibility to continue after fixing the issue, without starting over from scratch. Such as what happens when you run the installer with the docking station already plugged in. The test for this condition could come way earlier in the process. And may be assorted with a prompt to remove it, and press return to continue after removing it. Failure to download required files (see network issues above) seems to be unhandled as well, it just continues without, leaving an incomplete installation. Only solution is to start over from scratch. Error handling should not be a luxury...

And what's with these iffy bit.ly URLs? This is not twitter where you're limited to 160 characters (or whatever), and nobody is going to need to type them in manually (we have copy-paste). A readable URL, on the other hand, would build trust.

Thanks

AlainKnaff, 2020-08-08:
Found the reason: somehow the libevdi.so that ended up in /opt/displaylink/ was version 1.8.0, whereas the DisplayLinkManager (same directory) wanted 1.7.0.

I compared both with the version that I had in the directory from where I had run the displaylink-debian installer: the 2 files in /opt/displaylink were identical with those in the build directory:

- /opt/displaylink/libevdi.so was identical to ./evdi/library/libevdi.so and ./5.3.1/displaylink-driver-5.3.1/x64-ubuntu-1604/libevdi.so
- /opt/displaylink/DisplayLinkManager was identical to ./5.3.1/displaylink-driver-5.3.1/x64-ubuntu-1604/DisplayLinkManager

After replacing the libevdi.so from another installer which happened to have the required version 1.7.0, everything worked all right after a reboot. The xrandr --setprovideroutputsource 4 0 and xrandr --setprovideroutputsource 3 0 in /usr/share/sddm/scripts/Xsetup where still needed, but that's expected.

So, either the displaylink-debian installer was mis-packaged (mis-matched versions), or something bad happened when it downloaded further stuff from the Internet (maybe I was unlucky enough that versions on download sites changed somewhere in between my multiple attempts?)

The only way to find out about these mismatched version was to run DisplayManager under ltrace, and observe how it prepared error messages using memmove and memcpy, only to not write them out to anywhere... If these had actually been written anywhere (private logfile, syslog, systemctl status, ...) I would have picked up on this *much* earlier.

Also, the fact that the evdi.ko made available 4 graphic card devices even though it couldn't actually even communicate with the DisplayLink hardware was way confusing, misdirecting my attention to physical detection of monitors, rather than something like this.

I know, both of these debugging issues (absent error output, phantom devices without even communicating) is not you, but "upstream" (i.e. DisplayLink themselves), but if ever you communicate with them about future direction of usability, this might be something to bring up to them...

(I'm leaving the issue open for now, as the packaging issues are still relevant, even though I eventually got it to work.)

Thanks

rdmtinez, 2020-08-10:
Can you explain how you found out what DisplayLinkManager 'wanted'?, I'm having a similar issue to you and this might help me. Thanks in advance!

AlainKnaff, 2020-08-10:
I run the DisplayLinkManager under ltrace:

systemctl stop displaylink-driver
ltrace -s 255 -o mylog.txt /opt/displaylink/DisplayLinkManager

Then I looked for memmove lines in the ltrace log:

grep 'memmove' ltrace-logg  | less

==> you find that it uses memmove and memcpy to build its log and error messages (even though eventually they are printed anywhere...)

Currently I get something as the following:

memmove(0x1a95ab0, "00:00:00.074\t140060796843264:DisplayLinkMana*_7f62720af500\tDisplayLinkManager\t00007FFFD68C97B0\tALWAYS\t", 102) = 0x1a95ab0
memmove(0x1a95b90, "00:00:00.074\t140060796843264:DisplayLinkMana*_7f62720af500\tDisplayLinkManager\t00007FFFD68C97B0\tALWAYS\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/core/sys/libutils/Logger.cpp", 201) = 0x1a95b90
memmove(0x1a95ca1, "Started at ", 11)                                                                              = 0x1a95ca1
memmove(0x1a95cac, "Sat Aug  8 10:23:03 2020\n", 25)                                                               = 0x1a95cac
memmove(0x1a8f710, "\320\233\214\326\377\177\0\0", 8)                                                              = 0x1a8f710
memmove(0x1a97870, "00:00:00.435\t140060796843264:DisplayLinkMana*_7f62720af500\tDLM main\t00007FFFD68C9B80\tALWAYS\t", 92) = 0x1a97870
memmove(0x1a96d20, "00:00:00.435\t140060796843264:DisplayLinkMana*_7f62720af500\tDLM main\t00007FFFD68C9B80\tALWAYS\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/app/DisplayLinkManager/DisplayLinkManager.cpp", 222) = 0x1a96d20
memmove(0x1a96e02, "DisplayLinkManager PID(", 23)                                                                  = 0x1a96e02
memmove(0x1a96e19, "9349", 4)                                                                                      = 0x1a96e19
memmove(0x1a96e1d, ") start", 7)                                                                                   = 0x1a96e1d
memmove(0x1a97870, "00:00:00.461\t140060796843264:DisplayLinkMana*_7f62720af500\tDLM main\t00007FFFD68C9B80\tALWAYS\t", 92) = 0x1a97870
memmove(0x1a96d20, "00:00:00.461\t140060796843264:DisplayLinkMana*_7f62720af500\tDLM main\t00007FFFD68C9B80\tALWAYS\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/app/DisplayLinkManager/DisplayLinkManager.cpp", 222) = 0x1a96d20
memmove(0x1a96e02, "System uptime ", 14)                                                                           = 0x1a96e02
memmove(0x1a96e10, "Hours : 8 Minutes : 44 Seconds : 34", 35)                                                      = 0x1a96e10
memmove(0x1a9c6d0, "00:00:01.308\t140060796843264:DisplayLinkMana*_7f62720af500\t\t00007FFFD68CA588\tDEBUG\t", 83) = 0x1a9c6d0
memmove(0x1a95b90, "00:00:01.308\t140060796843264:DisplayLinkMana*_7f62720af500\t\t00007FFFD68CA588\tDEBUG\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libPOSIXDLM/DisplayLinkApp.cpp", 201) = 0x1a95b90
memmove(0x1a95c5d, "DisplayLinkApp::run ", 20)                                                                     = 0x1a95c5d
memmove(0x1a9c6d0, "00:00:01.755\t140060796843264:DisplayLinkMana*_7f62720af500\tEvdiLibrary\t0000000001A8F188\tINFO\t", 93) = 0x1a9c6d0
memmove(0x1a984e0, "00:00:01.755\t140060796843264:DisplayLinkMana*_7f62720af500\tEvdiLibrary\t0000000001A8F188\tINFO\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libEvdiDriver/EvdiLibrary.cpp", 210) = 0x1a984e0
memmove(0x1a985b7, "Loaded", 6)                                                                                    = 0x1a985b7
memmove(0x1a9c6d0, "00:00:01.810\t140060796843264:DisplayLinkMana*_7f62720af500\tEvdiLibrary\t0000000001A8F188\tINFO\t", 93) = 0x1a9c6d0
memmove(0x1a984e0, "00:00:01.810\t140060796843264:DisplayLinkMana*_7f62720af500\tEvdiLibrary\t0000000001A8F188\tINFO\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libEvdiDriver/EvdiLibrary.cpp", 210) = 0x1a984e0
memmove(0x1a985b7, "Initialized ", 12)                                                                             = 0x1a985b7
memmove(0x1a985c3, "successfully", 12)                                                                             = 0x1a985c3
memmove(0x1aa9800, "lo\0\0\0\0\0\0\0\0\0\0\0\0\0\0\002\0\0\0\177\0\0\001\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0enp0s31f6\0\0\0\0\0\0\0\002\0\0\0\300\250\2622\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4000) = 0x1aa9800
memmove(0x1a9c6d0, "00:00:03.812\t140060796843264:DisplayLinkMana*_7f62720af500\tDirectorImpl\t0000000001A980C8\tDEBUG\t", 95) = 0x1a9c6d0
memmove(0x1a984e0, "00:00:03.812\t140060796843264:DisplayLinkMana*_7f62720af500\tDirectorImpl\t0000000001A980C8\tDEBUG\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libPOSIXDLM/Director.cpp", 207) = 0x1a984e0
memmove(0x1a985b4, "DirectorImpl::start", 19)                                                                      = 0x1a985b4
memmove(0x1a9c6d0, "00:00:05.624\t140060796843264:DisplayLinkMana*_7f62720af500\t\t00007FFFD68C8C10\tDEBUG\t", 83) = 0x1a9c6d0
[... cut ...]
memmove(0x1a985b3, "Creating TiledDisplayGroupingStategy", 36)                                                     = 0x1a985b3

While I still had the problem, one of these said (quoting from memory, I don't have that initial logfile around any more): "DisplayLinkManager found libevdi.so version 1.8.0 compatibility version is 1.7.0"

i.e. it wanted 1.7.0, but unfortunately, somehow 1.8.0 had ended up in my /opt/displaylink/ directory. After copying an 1.7.0 version of libevdi.so from elsewhere, everything was fine and dandy :-)

AlainKnaff, 2020-08-10:
Actually, now that I'm taking a closer look at the libevdi sources, I see that the memmove'd message that I saw came from libevdi itself, rather than DisplayLinkManager. And it was complaining, in its bad chenglish "Doesn't match LibEvdi compatibility one (%d.%d.%d)", that libevdi.so was incompatible with the evdi kernel module...

rdmtinez, 2020-08-10:
Thanks for your help! I'm running it now, I'll come and update when I get the chance :+1:

rdmtinez, 2020-08-10:
I'm getting this message: 

memcpy(0x1aa2e10, "/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libEvdiDriver/EvdiApiFactory.cpp", 120) = 0x1aa2e10
_ZdlPv(0x1aa2d70, 0x1aa2d80, 120, 0x1aa2e88)     = 0
_ZdlPv(0x1aa2cd0, 0x1a79050, 0x1a79010, 0x1aa2e88) = 0
_Znwm(217, 0x1aa2bf1, 120, 0x1aa2b90)            = 0x1a9d5a0
memmove(0x1a9d5a0, "00:00:03.315\t140582061176832:DisplayLinkMana*_7fdbcfd17800\tEvdiApiFactory\t00007FFCFF6CE288\tERROR\t", 97) = 0x1a9d5a0
memcpy(0x1a9d601, "/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libEvdiDriver/EvdiApiFactory.cpp", 120) = 0x1a9d601
_ZdlPv(0x1aa2b90, 0x1aa2e10, 120, 0x1aa2b90)     = 0
_ZdlPv(0x1aa2e10, 0x1a79038, 0x4613fa, 2)        = 0
_Znwm(109, 0x1a79040, 0x1a79010, 1)              = 0x1aa2e10
_Znwm(434, 0x1a9d679, 217, 0x1a9d5a0)            = 0x1a9d3e0
memmove(0x1a9d3e0, "00:00:03.315\t140582061176832:DisplayLinkMana*_7fdbcfd17800\tEvdiApiFactory\t00007FFCFF6CE288\tERROR\t/home/jenkins/jobs/SW_Linux_PROD_branches_posix_5.4/workspace/src/nivo/docking/posix/tk/libEvdiDriver/EvdiApiFactory.cpp", 217) = 0x1a9d3e0
memcpy(0x1a9d4b9, ":17\tlibevdi.so cannot be loaded. Error: libevdi.so: cannot open shared object file: No such file or directory", 109) = 0x1a9d4b9

I guess it can't access the file, quite strange.

AlainKnaff, 2020-08-10:
>  memcpy(0x1a9d4b9, ":17\tlibevdi.so cannot be loaded. Error: libevdi.so: cannot open shared object file: No such file or directory", 109) = 0x1a9d4b9
> I guess it can't access the file, quite strange.

Indeed. Normally, the libevdi.so library is supposed to live in the /opt/displaylink/ directory, the same one where DisplayLinkManager lives.
It might be interesting to run DisplayLinkManager under strace.
Strace is the same thing as ltrace, but for system calls rather than library calls. This  _includes_ the loading of libraries. It might be interesting to see where DisplayLinkManager is looking for libevdi.so (normally it tries several places in succession, including the above-mentioned /opt/displaylink/ directory), and if it is found, whether it gets back an error code from the OS (permission, maybe). Or, if it can be read all right, this might indicate that the file is corrupted (it can be read, but the dynamic loader cannot make any sense of it as a library).

kennethtran93, 2020-09-10:
PR #457 definitely fixed my issue - also most likely the incorrect evdi version prior to this pr.

Barabazs, 2020-09-10:
Thanks @AlainKnaff

closed 2020-09-10 by Barabazs