AdnanHodzic/displaylink-debian #452: External monitor not working

ITA003, 2020-08-07:
Hi,
I'm upgraded to Mint 20 and reinstall the displaylink-debian tool and my external monitor stop working

My configuration is with two external monitor, one connected by HDMI and one connected with Displaylink.

With Mint 19.3 works all fine.

This is the debug output:

```
--------------- Linux system info ----------------

Distro: Linuxmint
Release: ulyana
Kernel: 5.4.0-42-generic

---------------- DisplayLink info ----------------

Driver version: 1.8.0
5.1.26
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: GT2
VGA: Intel Corporation Skylake GT2 [HD Graphics 520] (rev 07)
VGA (3D): 
X11 version: 1.20.8-2ubuntu2.2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x44 cap: 0x9, Source Output, Sink Offload crtcs: 3 outputs: 2 associated providers: 4 name:modesetting
Provider 1: id: 0x147 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x125 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x103 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0xe1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

-------------------------------------------------------------------
```

Initially the etc/X11/xorg.conf.d/20-displaylink.conf file doesn't exists, i created it after some search in Internet and contain:

```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

```

can you help me to make the external monitor works

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

ITA003, 2020-09-07:
Just updated and reinstalled. Works fine!

closed 2020-09-08 by Barabazs