AdnanHodzic/displaylink-debian #453: [Buster]No protocol Specified

magic-debian, 2020-08-11:
Hello All, 

A friend of mine recently gave me an Asus MB169B+ screen, i thought in the beginning it was just a matter of drivers to make it work on Debian so I downloaded the drivers from Asus's website (https://www.asus.com/Monitors/MB169BPlus/HelpDesk_Download/) : 

 `chmod +x displaylink.sh` 
then 
`# ./displaylink.sh` (yes i did as a root user, my bad)

Obviously it did not work, so I pulled this git, tried using the displaylink-debian.sh (as sudo) but i get the following mistake :+1: 
`sudo ./displaylink-debian.sh --debug
No protocol specified
Can't open display :0`

(That's it, no additional information provided like I can see people commenting with)

- I did follow all the post install instructions 

- I did create the file (manually)

`cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
`

- I did update Xorg

- I am using the latest version :+1: 

`sudo git checkout master && git log
Already on 'master'
Your branch is up to date with 'origin/master'.
commit ffe9791e78bc5ef925ce1f476a0fab51c5a23ee6 (HEAD -> master, origin/master, origin/HEAD)
Author: Barabas <31799121+Barabazs@users.noreply.github.com>
Date:   Wed Aug 5 17:30:02 2020 +0200

    Correct typo in Mint platform (#450)
`

I also uninstalled the Asus driver : 

`sudo displaylink-installer uninstall`

But got exactly the same issue of "no protocol specified" when trying to run displaylink-debian

Does any one has any idea new step I could do to solve my problem ?

jarandacu, 2020-08-26:
Same error here !

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest might resolve your issue. #457

Barabazs, 2020-09-06:
How are you running the installer? (directly on the machine, ssh, ...)
What's the output of `uname -a`

jackHalliday, 2020-09-06:
For me this was due to using Wayland rather than X. Aparantly recent versions of Debian default to Wayland with Gnome?

magic-debian, 2020-09-08:
So funny story... since it wasn't working for me I gave the screen to a friend of mine who runs windows on his laptop... (worked like a charm btw)

Sorry people, I won't be able to test anything ^^

closed 2020-09-08 by Barabazs

CmdrBugbear, 2020-09-11:
I'd like to respectfully reopen this issue as I'm getting the same symptoms.

I cloned this repository a couple of days ago, opened a root terminal

No protocol specified
Can't open display :0

I'm completely new to attempting to connect my debian laptop to a docking station, and apart from attempting to install the DisplayLink driver for ubuntu, I haven't deliberately configured this laptop at all.

Barabazs, 2020-09-11:
@CmdrBugbear 
I have opened a new issue for you. Let's continue there.