AdnanHodzic/displaylink-debian #456: External monitor connected via i-tec USB 3.0 dock not working

jdybdahl, 2020-08-29:
The monitor works fine via the i-tec dock with Surface 3 Pro on Ubuntu 20 and Windows.
However, the external monitor doesn't work on Debian 10 on my two main computers: hp Elitebook 8570p and a brand-new Lenovo e480. I connect with Gnome on Xorg on both. The dock happily took over audio, ethernet etc.
I have tried with two different monitors.

**** hp Elitebook 8570p:
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xd2 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 1 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

**** Lenovo e480:
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xce cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

**** dmesg | grep Display: on both computers complain about this:

[   16.976907] usb 2-1.1: Manufacturer: DisplayLink
[   17.561276] usb 2-1.1: [15] FU [DisplayLink Audio Playback Volume] ch = 6, val = -8176/0/16

**** xrandr report on both
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)

**** lsusb happily finds the dock Bus 004 Device 003: ID 17e9:4307 DisplayLink
and gives the usual l-o-n-g report. Anyone interested in that?

I have uninstalled / rebooted / installed displaylink-debian several times.

I have tried entering the two elsewhere suggested /etc/X11/xorg.conf.d/20-displaylink.conf files. They were ignored.

Any suggestions for how to progress?

jdybdahl, 2020-09-02:
For the record: J removed Ubuntu from one of the Microsoft Surface 3 Pros and installed Debian 10 - something I had wanted to do for some time. As expected, the dock does not work any longer.. :(

Barabazs, 2020-09-05:
Please uninstall and reinstall with the latest version. The latest fixes should resolve your issue. #457

jdybdahl, 2020-09-05:
Great to hear that there is work being done on this!

Alas, no, it is not working on my Surface Pro 3.
I use the command: xrandr --setprovideroutputsource 1 0 && xrandr --output eDP-1 --primary --output DVI-I-4-4 --auto --right-of eDP-1

The debug on the Surface Pro 3 is:
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Integrated
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.20.4-1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

I will try on the other machines tomorrow.

Barabazs, 2020-09-05:
Did you uninstall everything with displaylink-debian.sh ?

jdybdahl, 2020-09-05:
It didn't work on the HP 8570p either. :(

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xd2 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 1 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

jdybdahl, 2020-09-05:
Yes, I uninstalled, rebooted, installed, rebooted, plugged the dock in and executed the command.

Barabazs, 2020-09-05:
Can you post the output of both the uninstall and install please?
Evdi 1.8.0 shouldn't be installed at this point.

jdybdahl, 2020-09-05:
I uninstalled and now it fails:

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-10-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

rm: cannot remove '/lib/modules/4.19.0-10-amd64/build/Kconfig': No such file or directory
jorgen@saturn:~/displaylink$

Barabazs, 2020-09-05:
What does `dkms status` give?

jdybdahl, 2020-09-05:
I uninstalled and now it fails.
It downloads the newest driver automatically?

--------------------------- displaylink-debian -------------------------------

DisplayLink driver installer for Debian and Ubuntu based Linux distributions:

* Debian, Ubuntu, Elementary OS, Mint, Kali, Deepin and many more!
* Full list of all supported platforms: http://bit.ly/2zrwz2u
* When submitting a new issue, include Debug information

Options:

[I]nstall
[D]ebug
[R]e-install
[U]ninstall
[Q]uit

Select a key: [i/d/r/u/q]: u

-------------------------------------------------------------------

Checking dependencies

unzip is installed
linux-headers-4.19.0-10-amd64 is installed
dkms is installed
lsb-release is installed
linux-source is installed
x11-xserver-utils is installed
wget is installed
libdrm-dev is installed
git is installed

Platform requirements satisfied, proceeding ...

-------------------------------------------------------------------

Uninstalling ...

rm: cannot remove '/lib/modules/4.19.0-10-amd64/build/Kconfig': No such file or directory
jorgen@saturn:~/displaylink$

jdybdahl, 2020-09-05:
jorgen@saturn:~/displaylink$ sudo dkms status
evdi, 1.8.0, 4.19.0-10-amd64, x86_64: installed

http://xorgenism.blogspot.com/
    While money can't buy happiness, it certainly lets you choose your own form of misery.

Barabazs, 2020-09-05:
`sudo dkms remove evdi/1.8.0 --all`
Reboot
`dkms status`

jdybdahl, 2020-09-05:
jorgen@saturn:~/prj$ sudo dkms status
[sudo] password for jorgen: 
jorgen@saturn:~/prj$

jdybdahl, 2020-09-05:
No output after that.

I rebooted and installed successfully.
It still does not work.

HP 8570p:

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.8.0
DisplayLink service status: up and running
EVDI service version: 1.8.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xd2 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 1 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

.
You didn't answer my question: does the script download the latest stuff from displaylink?

Barabazs, 2020-09-05:
Did your reuse your existing script or did you download it again as requested?
The script downloads a specific evdi version.

jdybdahl, 2020-09-06:
Ah, I shouldn't test things while exhausted! I did what I had been doing ad nauseam: uninstalled and installed displaylink. Sorry!  I should have read all of it and seen that you were working on the scripts and not on the DisplayLink driver.

Anyway: now I lost my providers:

HP 8570p
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x4c cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 9 associated providers: 0 name:Intel

-------------------------------------------------------------------

I had tried setting the /etc/X11/xorg.conf.d/20-displaylink.conf to Intel without luck before.

jdybdahl, 2020-09-06:
I erased the etc/X11/xorg.conf.d/20-displaylink.conf and rebooted. I got my providers back but with the usual result.

jorgen@saturn:~/prj$ two
>>> ERROR: DVI-I-4-4 not connected <<<
*** Command:
xrandr --setprovideroutputsource 1 0 && xrandr --output LVDS-1 --primary --output *ERR* --auto --right-of LVDS-1
jorgen@saturn:~/prj$

Barabazs, 2020-09-06:
Try with this config in  /etc/X11/xorg.conf.d/20-displaylink.conf and reboot.
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

jdybdahl, 2020-09-06:
I get my providers but still no connection to DVI-I-4-4. :(
I had also tried that before.

jdybdahl, 2020-09-06:
One thing: my ethernet which is plugged into the dock used to be called something like "displaylink ethernet". Now it is just "ethernet" (just like if I plug the cable directly into the laptop).  Also, a green light has gone off on the dock (that light is as far as I can see not mentioned in the manual). This said: the ethernet works fine: I still get the 100 Mbps internet I pay for.

kngharv, 2020-09-22:
Would u switch to wayland and see if the external monitor detected?

jdybdahl, 2020-09-23:
Displaylink only create providers with X11, but OK:
jorgen@saturn:~/prj$ xrandr
Screen 0: minimum 320 x 200, current 1600 x 900, maximum 8192 x 8192
XWAYLAND0 connected 1600x900+0+0 (normal left inverted right x axis y axis) 350mm x 190mm
   1600x900      59.95*+
jorgen@saturn:~/prj$ 

http://xorgenism.blogspot.com/
    While money can't buy happiness, it certainly lets you choose your own form of misery.

kngharv, 2020-09-23:
Wayland worked for me:

```
$ xrandr
Screen 0: minimum 16 x 16, current 4640 x 2560, maximum 32767 x 32767
XWAYLAND0 connected 1440x2560+3200+0 (normal left inverted right x axis y axis) 340mm x 600mm
   1440x2560     59.93*+
XWAYLAND2 connected 1920x1080+1280+1078 (normal left inverted right x axis y axis) 340mm x 190mm
   1920x1080     59.96*+
XWAYLAND3 connected 1280x1024+0+1078 (normal left inverted right x axis y axis) 380mm x 300mm
   1280x1024     59.89*+
```

I am on Ubuntu/20.04 using 5.4.x generic kernel and for some reason the displaylink driver is older than the one you are using.  Mine is 1.7.0 instead of 1.8.x

what does your computer say when you type this:

`ls /sys/class/drm/*/status | xargs -I {} -i bash -c "echo -n {}: ; cat {}"`

would you provide output of this as well:
`
inxi -G`

jdybdahl, 2020-10-01:
Thank you for persuing this problem and sorry about the delay!
Am I doing this right?

This is the stuff I had before I changed to Xwayland:

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Gen
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
cat: /etc/X11/xorg.conf.d/20-displaylink.conf: No such file or directory
Contents:
 

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xd2 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 8 associated providers: 0 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

jorgen@saturn:~/prj$ xrandr
Screen 0: minimum 320 x 200, current 1600 x 900, maximum 8192 x 8192
LVDS-1 connected primary 1600x900+0+0 (normal left inverted right x axis y axis) 345mm x 194mm
   1600x900      60.03 +  59.99*   59.94    59.95    59.82    40.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
VGA-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
HDMI-3 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DVI-I-4-4 disconnected (normal left inverted right x axis y axis)

I then changed to Xwayland and rebooted and like I said, DisplayLink does not like Xwayland on Debian:

jorgen@saturn:~/displaylink$ sudo ./displaylink-debian.sh
No protocol specified
Can't open display :0

jorgen@saturn:~/displaylink$ ls /sys/class/drm/*/status | xargs -I {} -i bash -c "echo -n {}: ; cat {}"
/sys/class/drm/card0-DP-1/status:disconnected
/sys/class/drm/card0-DP-2/status:disconnected
/sys/class/drm/card0-DP-3/status:disconnected
/sys/class/drm/card0-HDMI-A-1/status:disconnected
/sys/class/drm/card0-HDMI-A-2/status:disconnected
/sys/class/drm/card0-HDMI-A-3/status:disconnected
/sys/class/drm/card0-LVDS-1/status:connected
/sys/class/drm/card0-VGA-1/status:disconnected
/sys/class/drm/card1-DVI-I-1/status:connected
/sys/class/drm/card2-DVI-I-2/status:disconnected
/sys/class/drm/card3-DVI-I-3/status:disconnected
/sys/class/drm/card4-DVI-I-4/status:disconnected

jorgen@saturn:~/displaylink$ inxi -G
Graphics:
  Device-1: Intel 3rd Gen Core processor Graphics driver: i915 v: kernel 
  Display: wayland server: X.Org 1.20.4 driver: modesetting 
  unloaded: fbdev,vesa resolution: 1600x900~60Hz 
  OpenGL: renderer: Mesa DRI Intel Ivybridge Mobile v: 4.2 Mesa 18.3.6 

http://xorgenism.blogspot.com/
    While money can't buy happiness, it certainly lets you choose your own form of misery.

kngharv, 2020-10-02:
`jorgen@saturn:~/displaylink$ ls /sys/class/drm/*/status | xargs -I {} -i bash -c "echo -n {}: ; cat {}"
/sys/class/drm/card0-DP-1/status:disconnected
/sys/class/drm/card0-DP-2/status:disconnected
/sys/class/drm/card0-DP-3/status:disconnected
/sys/class/drm/card0-HDMI-A-1/status:disconnected
/sys/class/drm/card0-HDMI-A-2/status:disconnected
/sys/class/drm/card0-HDMI-A-3/status:disconnected
/sys/class/drm/card0-LVDS-1/status:connected
/sys/class/drm/card0-VGA-1/status:disconnected
/sys/class/drm/card1-DVI-I-1/status:connected
/sys/class/drm/card2-DVI-I-2/status:disconnected
/sys/class/drm/card3-DVI-I-3/status:disconnected
/sys/class/drm/card4-DVI-I-4/status:disconnected`

1. It seems that you have two monitors.  is that correct?

2. what kind of desktop are you using?  

3. do you still suffer the problem of not having external monitor detected?

jdybdahl, 2020-10-03:
-    
>>> It seems that you have two monitors. is that correct?

   - Yes.   

   -    
>>> what kind of desktop are you using?

   - I built it myself and the desktop doesn't matter.    

   - The only important computers are two of my laptops: a HP Elitebook 8570p and a brand-new Lenovo e480, both running Debian 10.    

   -    
>>> do you still suffer the problem of not having external monitor detected?

   - Yes.

http://xorgenism.blogspot.com/
    While money can't buy happiness, it certainly lets you choose your own form of misery.

kngharv, 2020-10-08:
Your set up is a lot more complicated.  It seems you are running wayland underneath everything and Xwayland on top of wayland to get around other components which requires X11 server.

Because of it, it is a bit hard to debug.

I have a pure wayland desktop (wayland + gnome 3.18) and displaylink detected monitors automatically.

I also have a pure X11 desktop (X11 + xfce + compton just for kicks) and I finally managed to get my X11 environment detect the external monitor, you can read it here:

https://github.com/AdnanHodzic/displaylink-debian/issues/469

Good luck

jdybdahl, 2020-10-09:
1) Re. Wayland/xwayland: I just uncommented the usual "#WaylandEnable=false"I have anyway set both laptops to X11.

2) Harvey, your description in the #469 link is excellent! On my laptops this works:
The USB to the dock must be unplugged so it re-initialises.
Plug it in and execute a script with the two outputsources that work for you; in my case

xrandr --setprovideroutputsource 1 0xrandr --setprovideroutputsource 4 0

Excellent! Now it works on the two laptops that matter to me! Thank you so much.
Also many thanks to Barrabas for the evdi 1.7 etc work and, yes,Section "OutputClass"
	Identifier "DisplayLink"
	MatchDriver "evdi"
	Driver "modesetting"
	Option  "AccelMethod" "none"
EndSection
works for me too.

Adrian, one problem in the displaylink-debian stuff:On the Lenovo e480, I erased all the displaylink stuff, rebooted and reinstalled with
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh 
wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/evdi.sh
chmod +x displaylink-debian.sh evdi.sh
sudo ./displaylink-debian.sh

It then complained it was missing the resources/displaylink-installer.patch.I fetched it from the HP laptop and all is now well.

I am ecstatic! I lost my expensive door stopper and gained a dock!  :)