AdnanHodzic/displaylink-debian #458: Monitor not detected using debian testing

OParczyk, 2020-09-09:
Hi! I just got two new DisplayLink-Monitors, for which I tried to install drivers using this tool today.
I was however unsuccessful, and the external monitors both aren't being detected.
I tried installation (followed by reboot, test, uninstall, reboot) on Kernel versions 4.19 and 5.7, both yield the same result.

During installation (regardless of Kernel) I got the following error-esque block:
```
-------------------------------------------------------------------

Preparing for install

Archive:  DisplayLink_Ubuntu_5.3.1.zip
  inflating: 5.3.1/displaylink-driver-5.3.1.34.run  
Verifying archive integrity... All good.
Uncompressing DisplayLink Linux Driver 5.3.1.34

-------------------------------------------------------------------
Paketlisten werden gelesen...
Abhängigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
libdrm-dev ist schon die neueste Version (2.4.102-1).
libelf-dev ist schon die neueste Version (0.180-1+b1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 3 nicht aktualisiert.
https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch
patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Verzeichnis „/home/user/dislink/displaylink-debian/evdi/module“ wird betreten
make -C /lib/modules/5.7.0-3-amd64/build M=$PWD
make[2]: Verzeichnis „/usr/src/linux-headers-5.7.0-3-amd64“ wird betreten
  AR      /home/user/dislink/displaylink-debian/evdi/module/built-in.a
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_drv.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_modeset.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_connector.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_encoder.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_main.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_fb.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_gem.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_painter.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_params.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_cursor.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_debug.o
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi_ioc32.o
  LD [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi.o
  MODPOST 1 modules
  CC [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi.mod.o
  LD [M]  /home/user/dislink/displaylink-debian/evdi/module/evdi.ko
make[2]: Verzeichnis „/usr/src/linux-headers-5.7.0-3-amd64“ wird verlassen
make[1]: Verzeichnis „/home/user/dislink/displaylink-debian/evdi/module“ wird verlassen
CFLAGS="-I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers " make -C library 
make[1]: Verzeichnis „/home/user/dislink/displaylink-debian/evdi/library“ wird betreten
cc -I../module -std=gnu99 -fPIC -I../module -Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers    -c -o evdi_lib.o evdi_lib.c
cc evdi_lib.o -shared -Wl,-soname,libevdi.so.0 -o libevdi.so.1.7.0 -lc -lgcc 
cp libevdi.so.1.7.0 libevdi.so
make[1]: Verzeichnis „/home/user/dislink/displaylink-debian/evdi/library“ wird verlassen
patching file 5.3.1/displaylink-driver-5.3.1/displaylink-installer.sh

Installing driver version: 5.3.1

DisplayLink Linux Software 5.3.1.34 install script called: install
Distribution discovered: Debian GNU/Linux bullseye/sid

Installing

[ Installing EVDI ]
[[ Installing EVDI DKMS module ]]

Creating symlink /var/lib/dkms/evdi/1.7.0/source ->
                 /usr/src/evdi-1.7.0

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=5.7.0-3-amd64 all INCLUDEDIR=/lib/modules/5.7.0-3-amd64/build/include KVERSION=5.7.0-3-amd64 DKMS_BUILD=1....
cleaning build area...

DKMS: build completed.

evdi.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.7.0-3-amd64/updates/dkms/

depmod....

DKMS: install completed.
[[ Installing module configuration files ]]
[[ Installing EVDI library ]]
make: Nothing to be done for 'default'.
[ Installing x64-ubuntu-1604/DisplayLinkManager ]
[ Installing libraries ]
[ Installing firmware packages ]
[ Installing licence file ]
[ Adding udev rule for DisplayLink DL-3xxx/4xxx/5xxx/6xxx devices ]
[ Adding upstart and powermanager sctripts ]
```

running your script with "--debug" flag yields:

```
 -------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Debian
Release: bullseye
Kernel: 5.7.0-3-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
snd_hda_intel
Subsystem: Core
VGA: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
NVIDIA Corporation GF119M [Quadro NVS 4200M] (rev ff)
VGA (3D): 
X11 version: 1.20.8-2
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0x43 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x137 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x115 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xf3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xd1 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

------------------------------------------------------------------- 
```

the above is the same when using Kernel 4.19. I am at a total loss here, could someone help me out?

mlapan4, 2020-09-17:
I'm having the same issue using Kernel 4.19.0.  It seems like the DisplayLink Driver Version should be 5.3.1 or 5.3.xxx, but I am getting the same thing where it matches the EVDI service version.  Here is the output on my machine of sudo ./displaylink-debian.sh --debug:

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-10-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation HD Graphics 620 (rev 02)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-intel.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 2 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

kngharv, 2020-09-22:
I had the same problem with my Ubuntu/20.04.  I never resolve the monitor not detecting issue when I uses X11.

When I switched to wayland,  the monitor is properly detected.   

Would you switch to wayland and see if the external monitor can be detected by the displaylink driver?  If the displaylink driver detect monitor consistently in wayland but not in X11, it may shed some light on where the problem might be.

kngharv, 2020-10-08:
https://github.com/AdnanHodzic/displaylink-debian/issues/469

OParczyk, 2020-10-09:
Thank you! I will try once I get my hands on the monitors again. Wayland however is no option for me as I rely on X11 and bumblebee.

mlapan4, 2020-10-15:
This instantly worked for me, thanks so much!

closed 2020-10-19 by Barabazs

OParczyk, 2020-10-21:
Not quite abandoned, but #469 worked for me, leaving this here for future reference.
Thank you!