AdnanHodzic/displaylink-debian #459: blankscreen only cursor works, i915 debian buster (both stock and backport kernels).

rsisinni, 2020-09-11:
Hi I have really tried, before reporting this.  But symptoms all point to the pageflip issue, but the workarounds do not seem to work.  The behaviour is always the same.  No auto-detected montor, have to manual trigger.  And then only black output, and mouse cursor.  If I turn on SWCursor on the modesetting driver then I get only black.  I've tried with the default kernel, then I tried with back ported kernel as well. 

BTW I read the post install and I tried the two workarounds for this issue. 
Also I am running xorg not wayland

This is a displayport output, thats provider 4. I enable with
 xrandr  --setprovideroutputsource 4 0 - that does work to turn on the output.

Here is the latest config, and by no means is it the only one I tried.  
I have also tried disabling pagefip on the intel driver
diabling shadow buffer on modesetting 

--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.7.0-0.bpo.2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.0.0
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFile" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xee cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 1 name:Intel
Provider 1: id: 0xc5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xa3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x81 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

rsisinni, 2020-09-11:
Sorry here is the latest I noticed when I posted this one I have pagefile instead of pageflip.
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 5.7.0-0.bpo.2-amd64

---------------- DisplayLink info ----------------

Driver version: 1.0.0
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: 620
VGA: Intel Corporation UHD Graphics 620 (rev 07)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "DisplayLink"
    Driver      "modesetting"
    Option      "PageFlip" "false"
EndSection

-------------------- Monitors --------------------

Providers: number : 5
Provider 0: id: 0xee cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 5 associated providers: 1 name:Intel
Provider 1: id: 0xc5 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0xa3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x81 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

Barabazs, 2020-09-11:
Did you try this config yet? 
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

rsisinni, 2020-09-11:
Yes, list providers then only shows the first entry, which is the integrated intel graphics.

rsisinni, 2020-09-14:
Any other Ideas, I've gone back to older versions as well, with corresponding evdi.  And played with different versions of evdi as well as leaving the drivers version in place. The behavoiur is pretty much the same on all versions, except for the number of providers is less on older versions.  I also see in Xorg log with some configurations messages about ignoring evdi.0 evdi.1 etc

rsisinni, 2020-09-25:
I seem to have found something significant in the logs.

user.log:Sep 25 11:48:01 treborlt /usr/lib/gdm3/gdm-x-session[2223]: gbm: Last dlopen error: /usr/lib/dri/evdi_dri.so: cannot open shared object file: No such file or directory

There is no /usr/lib/dri directory at all, I did a find from the root of the system it dosn't seem to be there period.   I will change the the script to leave the artifacts to see if its there. after the build of evdi.

Barabazs, 2020-10-10:
Maybe try up- or downgrading xorg-server.

closed 2020-10-16 by Barabazs