AdnanHodzic/displaylink-debian #460: No Protocol Specified

Barabazs, 2020-09-11:
I'd like to respectfully reopen this issue as I'm getting the same symptoms.

I cloned this repository a couple of days ago, opened a root terminal

No protocol specified
Can't open display :0

I'm completely new to attempting to connect my debian laptop to a docking station, and apart from attempting to install the DisplayLink driver for ubuntu, I haven't deliberately configured this laptop at all.

_Originally posted by @CmdrBugbear in https://github.com/AdnanHodzic/displaylink-debian/issues/453#issuecomment-691016815_

CmdrBugbear, 2020-09-11:
...gathering troubleshooting information...

Debian version via lsb_release:
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

Barabazs, 2020-09-11:
Please give the output for:
`uname -a`
`echo $WAYLAND_DISPLAY`
`echo $DISPLAY`

CmdrBugbear, 2020-09-11:
I've run these commands as non-root.

(base) xxxxxx@xxxxxx:~/Documents/Programming/display_link/displaylink-debian$ uname -a
Linux xxxxxx 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux
(base) xxxxxx@xxxxxx:~/Documents/Programming/display_link/displaylink-debian$ echo $WAYLAND_DISPLAY
wayland-0
(base) xxxxxx@xxxxxx:~/Documents/Programming/display_link/displaylink-debian$ echo $DISPLAY
:0
(base) xxxxxx@xxxxxx:~/Documents/Programming/display_link/displaylink-debian$

Barabazs, 2020-09-11:
Please reformat your reply in a readable manner. (ie backticks etc)

CmdrBugbear, 2020-09-11:
uname -a
Linux apsl03 4.19.0-10-amd64 #1 SMP Debian 4.19.132-1 (2020-07-24) x86_64 GNU/Linux

echo $WAYLAND_DISPLAY
wayland-0

echo $DISPLAY
:0

(my apologies)

Barabazs, 2020-09-11:
Try `echo $XDG_SESSION_TYPE` please.

CmdrBugbear, 2020-09-11:
$XDG_SESSION_TYPE
wayland

Barabazs, 2020-09-11:
This installer doesn't support wayland. You'll have to use X.

closed 2020-09-11 by Barabazs

CmdrBugbear, 2020-09-11:
OK thanks for the quick response.

CmdrBugbear, 2020-09-11:
Minor update for anyone with the same issue - this link https://wiki.debian.org/Wayland gave me the info needed to disable Wayland.

Scroll down to the "Display Managers" section:

Display Managers

GDM3

GDM (GNOME Display Manager) will automatically use Wayland when supported, except when using the proprietary NVIDIA driver, in which case it will fall back to X11 due to instability.

To use the X11 backend by default, uncomment the WaylandEnable=false line in the /etc/gdm3/daemon.conf file.