AdnanHodzic/displaylink-debian #464: Unsupported platform on linux mint tricia

khru, 2020-09-25:
[image]

afabozzi, 2020-09-27:
I am having this same issue. Any help appreciated. Thanks.

afabozzi, 2020-09-27:
Turned out to be a typo in the script. Edit displaylink-debian.sh and on line 196 change the m to uppercase like so:

elif [ "$lsb" == "LinuxMint" ];

After doing that and running the script it worked however my laptop did freeze on the install. I waited a while with no change so I pressed Ctrl+Alt+F5 and logged into terminal and rebooted the machine:

sudo reboot now

After reboot my second screen is now working properly.

THANKS FOR THIS AWESOME SCRIPT GUYS!!!

AdnanHodzic, 2020-09-28:
@afabozzi please consider [creating a PR](https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) to merge these changes with master. Thanks!

Barabazs, 2020-10-07:
> @afabozzi please consider [creating a PR](https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request) to merge these changes with master. Thanks!

The script used to check for "LinuxMint" but some users reported that `lsb_release -is` was "Linuxmint". (ie https://github.com/AdnanHodzic/displaylink-debian/issues/448#issuecomment-668785550) The script was modified with this PR https://github.com/AdnanHodzic/displaylink-debian/pull/450

We could do a case insensitive comparison with grep, like this:
```diff
--- elif [ "$lsb" == "Linuxmint" ];
+++ elif echo $lsb | grep -qi "LinuxMint" ;
```

AdnanHodzic, 2020-10-10:
@Barabazs sounds like good idea to me. As I suggested previously, if someone could make a PR I'll be glad to review and merge with master.

PR #470 (2020-10-10): Perform case insensitive comparison of lsb_release for LinuxMint
Some users have reported that the output of  `lsb_release -is` was "Linuxmint" and others had "LinuxMint".

Closes #464
closed 2020-10-10 by AdnanHodzic