AdnanHodzic/displaylink-debian #465: MX-19.2_KDE_x64

TheBlogHouse, 2020-09-25:
Hey

Just trying to move over to Linux after many years trying different distros and settled on MX-19.2_KDE_x64 and using 5.8 kernel and wondering if your latest version should support this before I try debugging the issues I am having?

Thanks

kngharv, 2020-10-02:
Instead of solving the problem, have you consider to get around it?  I thinks for MX Linux, there is an boot option to use systemd instead of sysVinit.

Try that first. 

(i am trying to figure this out myself as we speak)

TheBlogHouse, 2020-10-02:
Yes I did but after many hours gave up, installed Kubuntu which has a newer version of Plasma by default, installed the latest 5.8.12 kernel using the great mainline app and the latest displaylink-driver-5.3.1.34 from Display link and all works well 👍

kngharv, 2020-10-02:
1. what do you mean by "latest displaylink-driver 5.3.1.34?"  did you use this installation script?  or you download the driver directly?
2.  are you using X11 or wayland?  I am having the hardest problem get my external monitor detected when I am using X11.  I have tried this displaylink-debian script on 3 different distros and none of them can detect external monitor when Xorg is used.

TheBlogHouse, 2020-10-03:
> 1. what do you mean by "latest displaylink-driver 5.3.1.34?"  did you use this installation script?  or you download the driver directly?
> 2. are you using X11 or wayland?  I am having the hardest [...]

I answered above but maybe see https://www.displaylink.org/forum/showpost.php?p=90773&postcount=106 for context

closed 2020-10-08 by Barabazs