AdnanHodzic/displaylink-debian #466: Debian 10 + Dynadock U3.0 - displays disconnected / not detected(?)

CmdrBugbear, 2020-10-04:
Hi,
After enabling X / disabling Wayland, I've been able to get the displaylink-debian.sh (install) script to execute successfully.

I rebooted after installation and displaylink-debian.sh --debug (see attached file) indicates that it is installed and running:
---------------- DisplayLink info ----------------

Driver version: 1.6.0+dfsg
1.7.0
DisplayLink service status: up and running
EVDI service version: 1.6.0

xrandr --listproviders outputs the following (I presume this means my monitors are being detected.  What does a working instance look like?):

xrandr --listproviders
Providers: number : 6
Provider 0: id: 0xfc cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 8 associated providers: 2 name:modesetting
Provider 1: id: 0xd2 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0xb0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x8e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x6c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x4a cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 6 outputs: 5 associated providers: 0 name:ATI FirePro (FireGL) Graphics Adapter @ pci:0000:01:00.0

I ran the xrandr --setprovideroutputsource commands.  These commands didn't have any apparent effect.  Are the parameters for this command dependant on the providers listed above?  If so, which should I have chosen?

When I run xrandr without parameters I get the following (my laptop lid is closed and I'm running of a screen connected via a single HDMI-out port, and my dock has two monitors connected, one via an HDMI port, and one via a DVI port):
xrandr
Screen 0: minimum 320 x 200, current 1920 x 1080, maximum 8192 x 8192
LVDS-1 connected (normal left inverted right x axis y axis)
   1920x1080     60.01 +  60.01    59.97    59.96    59.93    40.01  
   1680x1050     59.95    59.88  
   1600x1024     60.17  
   1400x1050     59.98  
   1600x900      59.99    59.94    59.95    59.82  
   1280x1024     60.02  
   1440x900      59.89  
   1400x900      59.96    59.88  
   1280x960      60.00  
   1440x810      60.00    59.97  
   1368x768      59.88    59.85  
   1360x768      59.80    59.96  
   1280x800      59.99    59.97    59.81    59.91  
   1152x864      60.00  
   1280x720      60.00    59.99    59.86    59.74  
   1024x768      60.04    60.00  
   960x720       60.00  
   928x696       60.05  
   896x672       60.01  
   1024x576      59.95    59.96    59.90    59.82  
   960x600       59.93    60.00  
   960x540       59.96    59.99    59.63    59.82  
   800x600       60.00    60.32    56.25  
   840x525       60.01    59.88  
   864x486       59.92    59.57  
   800x512       60.17  
   700x525       59.98  
   800x450       59.95    59.82  
   640x512       60.02  
   720x450       59.89  
   700x450       59.96    59.88  
   640x480       60.00    59.94  
   720x405       59.51    58.99  
   684x384       59.88    59.85  
   680x384       59.80    59.96  
   640x400       59.88    59.98  
   576x432       60.06  
   640x360       59.86    59.83    59.84    59.32  
   512x384       60.00  
   512x288       60.00    59.92  
   480x270       59.63    59.82  
   400x300       60.32    56.34  
   432x243       59.92    59.57  
   320x240       60.05  
   360x202       59.51    59.13  
   320x180       59.84    59.32  
VGA-1 disconnected (normal left inverted right x axis y axis)
HDMI-1 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 819mm x 460mm
   1920x1080     60.00*+  50.00    59.94  
   1920x1080i    60.00    50.00    59.94  
   1280x1024     75.02    60.02  
   1280x960      60.00  
   1360x768      60.02  
   1152x864      75.00  
   1280x720      60.00    50.00    59.94  
   1024x768      75.03    70.07    60.00  
   800x600       72.19    75.00    60.32    56.25  
   720x576       50.00  
   720x576i      50.00  
   720x480       60.00    59.94  
   720x480i      60.00    59.94  
   640x480       75.00    72.81    60.00    59.94  
   720x400       70.08  
DP-1 disconnected (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
HDMI-3 disconnected (normal left inverted right x axis y axis)
DP-2 disconnected (normal left inverted right x axis y axis)
DP-3 disconnected (normal left inverted right x axis y axis)
DVI-I-5-4 disconnected (normal left inverted right x axis y axis)
DVI-I-4-3 disconnected (normal left inverted right x axis y axis)

I have a couple of questions at this point:  
 - Is there a way to know whether my dock is actually supported?
 - How do I get the screens connected to the dock to be shown as connected?

[debug_20201004.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5323152/debug_20201004.txt)

Hardware configuration info:
(Not sure if this info is relevant but I'll provide it regardless)
Laptop is a 2012 Alienware M17x R3
VGA port has nothing connected
Mini-DP port has nothing connected
HDMI port is connected to a screen
USB3.0 port connected to Toshiba Dynadock U3.0
Dynadoc has HDMI and DVI ports, both connected to screens.

I've also attached the output of dmesg when plugging in the Dynadock.

[dmesg_20201004.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5323180/dmesg_20201004.txt)

kngharv, 2020-10-09:
try this:
https://github.com/AdnanHodzic/displaylink-debian/issues/469

CmdrBugbear, 2020-10-10:
OMG you're an f***ing legend (forgive the profanity).

I ran the command: xrandr --setprovideroutputsource n 0 (n times, where n is the number of providers).

Initially nothing happened, but then I realised the dock wasn't plugged in.  Pretty much straight away all my monitors woke up.  I'm excited because up until now I could only get this laptop to output display on up to two screens, no matter how many video out ports were pluged in.  Didn't know how to make it output to any more than that.

Thank you so much.

I'll drop a tip in the jar today.

closed 2020-10-10 by CmdrBugbear