AdnanHodzic/displaylink-debian #467: Displaylink doesn't install on Debian9. Stops with error messages concerning evdi_drv.c

d-hezel, 2020-10-06:
Hi there, 
I've been trying to install the Displaylink-Driver but I'm getting the error messages you see below. The installation stops unfinished. Driver is not installed. 
As I'm not really experienced using Linux I absolutely can't tell what the problem is nor find any suitable workaround for this on my own. A part of the messages is in german but I guess, the most important errors are in english. Please ask if you need any further information or want me to translate the german parts. If you need any logs or further stuff please consider me an absolute beginner and keep things simple... 

I'm running Debian 9, kernel version is 4.9.0-8-amd64, KDE-Plasma version 5.8.6. 

Thanks for your help! 

> CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Verzeichnis „/home/dominik/evdi/module“ wird betreten
make -C /lib/modules/4.9.0-8-amd64/build M=$PWD
make[2]: Verzeichnis „/usr/src/linux-headers-4.9.0-8-amd64“ wird betreten
  LD      /home/dominik/evdi/module/built-in.o
  CC [M]  /home/dominik/evdi/module/evdi_drv.o
/home/dominik/evdi/module/evdi_drv.c:49:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .fault = evdi_gem_fault,
           ^~~~~~~~~~~~~~
/home/dominik/evdi/module/evdi_drv.c:49:11: note: (near initialization for ‘evdi_gem_vm_ops.fault’)
/home/dominik/evdi/module/evdi_drv.c:86:12: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .unload = evdi_driver_unload,
            ^~~~~~~~~~~~~~~~~~
/home/dominik/evdi/module/evdi_drv.c:86:12: note: (near initialization for ‘driver.unload’)
/home/dominik/evdi/module/evdi_drv.c: In function ‘evdi_platform_probe’:
/home/dominik/evdi/module/evdi_drv.c:191:2: error: implicit declaration of function ‘drm_dev_put’ [-Werror=implicit-function-declaration]
  drm_dev_put(dev);
  ^~~~~~~~~~~
/home/dominik/evdi/module/evdi_drv.c: In function ‘evdi_platform_remove’:
/home/dominik/evdi/module/evdi_drv.c:201:2: error: implicit declaration of function ‘drm_dev_unplug’ [-Werror=implicit-function-declaration]
  drm_dev_unplug(drm_dev);
  ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
/usr/src/linux-headers-4.9.0-8-common/scripts/Makefile.build:309: die Regel für Ziel „/home/dominik/evdi/module/evdi_drv.o“ scheiterte
make[5]: *** [/home/dominik/evdi/module/evdi_drv.o] Fehler 1
/usr/src/linux-headers-4.9.0-8-common/Makefile:1546: die Regel für Ziel „_module_/home/dominik/evdi/module“ scheiterte
make[4]: *** [_module_/home/dominik/evdi/module] Fehler 2
Makefile:152: die Regel für Ziel „sub-make“ scheiterte
make[3]: *** [sub-make] Fehler 2
Makefile:8: die Regel für Ziel „all“ scheiterte
make[2]: *** [all] Fehler 2
make[2]: Verzeichnis „/usr/src/linux-headers-4.9.0-8-amd64“ wird verlassen
Makefile:71: die Regel für Ziel „module“ scheiterte
make[1]: *** [module] Fehler 2
make[1]: Verzeichnis „/home/dominik/evdi/module“ wird verlassen
Makefile:8: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
cp: der Aufruf von stat für 'library/libevdi.so.1.*' ist nicht möglich: Datei oder Verzeichnis nicht gefunden <

Barabazs, 2020-10-10:
Please post the output of `sudo ./displaylink-debian.sh --debug`

jdybdahl, 2020-10-11:
I can answer that for version 4.9.0-13-amd64. The debug gives no real info :(

"Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: No such file or directory"

It is not a problem for me as I can wipe the computer and put version 10 on it.
However, if you can fix this, I will wait and keep Debian 9.

d-hezel, 2020-10-11:
Correct! That's all... see below...

--------------------------------------------------------------
Starting Debug ...

Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y

Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y

cat: /sys/devices/evdi/version: Datei oder Verzeichnis nicht gefunden
---------------------------------------------------------------

Barabazs, 2020-10-11:
Displaylink dropped support for kernels < 4.15 DisplayLink/evdi@d242444

closed 2020-10-11 by Barabazs