AdnanHodzic/displaylink-debian #469: how i made displaylink detect my external monitor

kngharv, 2020-10-08:
I managed to make my displaylink hub detect my external monitor when X11 server is used.  Albeit, my hardware is simple with no discrete GPU.

I thought the problem lies upon my xorg setting, thus tried just about every xorg.conf file in the forum, didn't work.

The key, lies upon TWO commands in the documentation.

First command is the following:
`
xrandr --listproviders
`

Once you successfully installed the displaylink device driver, you should see "multiple lines"  indicate that there are "multiple source,"  each line represents some sort of "GPU."

Let say your computer have an Intel CPU and a Nvidia GPU card, you should expect to see TWO lines before you install the displaylink driver.   Once you install the displaylink driver, you would THINK the two lines will become THREE, as one additional line represents the displaylink GPU.

In my case, I don't have Nvidia nor AMD GPU, thus, I only had one single line before displaylink driver installed.

When I plug in my displaylink hub,  instead of two lines as I expected, I this instead:
```
xrandr --listproviders
Providers: number : 5
Provider 0: id: 0xcf cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 5 associated providers: 4 name:modesetting
Provider 1: id: 0xa7 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x85 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x63 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x41 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting

```

The first line was originally labelled "intel."  So, I know that thing is a real GPU.  But ALL the rest are displaylink "gpus?"

The second line in question is the following:
`
xrandr --setprovideroutputsource 1 0
`

I didn't know what this meant until I ran into some article in Arch Linux's forum.   This line somehow connect the real intel GPU to the display GPU.

My mistake is that I merely follow the document here, and I merely executed:
```

xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0

```
Since I actually have total of FOUR displaylink GPUs,  I really need to link ALL of them to the intel GPU.

```
xrandr --setprovideroutputsource 1 0
xrandr --setprovideroutputsource 2 0
xrandr --setprovideroutputsource 3 0
xrandr --setprovideroutputsource 4 0

```
it turned out that all my external monitors were linked to the "4th" displaylink gpu.  As soon as I execute `xrandr --setprovideroutputsource 4 0`, my computer detected external monitors... all two of them.

In short, 
1. execute  `xrandr --listproviders` 
2. count number of providers in your hub
3. execute ` xrandr --setprovideroutputsource ` n times
```
 xrandr --setprovideroutputsource 1 0
 xrandr --setprovideroutputsource 2 0
 xrandr --setprovideroutputsource 3 0
 .
 .
 .
 .
 xrandr --setprovideroutputsource n 0

```

where n is the number of additional lines appears when displaylink driver is installed.

Good luck.

jdybdahl, 2020-10-09:
Well done! This was the last step I needed to make it work! 
Thank you so much!

Jorgen

Barabazs, 2020-10-10:
Thank you for your contribution @kngharv !

robotard, 2020-10-13:
I'm actually having a bit of the same issue... I can get my Ryzen 7 laptop to do HDMI out separately to USB 3.0 DisplayLink... (Audio is a bitch on each restart... takes DisplayLink over HDMI)... 

Is there a way to Nvidia & DisplayLink to play nicely?

Here are my efforts from the above (but swear it's trying to Radeon Vega me instead of NVidia me...)

xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x27a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x258 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x236 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x214 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f2 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 1 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 2 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 3 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 4 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 5 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x1f2
  Serial number of failed request:  20
  Current serial number in output stream:  21
robsta@robstas-um433iq:~$ xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x27a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x258 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x236 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x214 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f2 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0

robotard, 2020-10-13:
Please ignore the strikethroughs hehe

jdybdahl, 2020-10-13:
Robotard, DisplayLink says re Ubuntu: "Closed source AMD/NVIDIA drivers are incompatible with DisplayLink driver. Please use open-source AMD/NVIDIA drivers instead."
https://support.displaylink.com/knowledgebase/articles/641668-known-issues-with-displaylink-ubuntu-support

robotard, 2020-10-13:
That’s strange. Working fine on Kernel 4.8.

robotard, 2020-10-13:
Apparently, there’s a patch that I’ll test later tonight

closed 2020-11-13 by Barabazs