AdnanHodzic/displaylink-debian #471: How to make displaylink-debian work on Microsoft Surface Pro 3

jdybdahl, 2020-10-11:
Please note that Microsoft Surface Pro 3 with Debian 10 gets this .conf file:

Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

Either comment the text out or erase it
      
> sudo nano /etc/X11/xorg.conf.d/20-displaylink.conf

or erase the file.

closed 2020-10-11 by Barabazs

jdybdahl, 2020-10-11:
Why "invalid"? It is a necessary step.
Shouldn't it be "informational"?

Barabazs, 2020-10-11:
Your post and intent is unclear and lacks context.

jdybdahl, 2020-10-11:
Really?  It clearly says that I used displaylink-debian on a Microsoft Surface Pro 3 and ended up with a .conf file that prevented my dock from connecting to my external screen.  

It is useful information to anyone trying to do the same. 
It is useful to you as you can advice people with a Surface Pro 3 correcty.
And it is useful for people who gets a bad .conf file on other computers as it tells them: try getting rid of it.

But have it your way.

Anyway, again thanks for your help with my i-tec dock! It is all working beautifully now.

jdybdahl, 2020-10-13:
Since you now have argued your side of the case, allow me to state my side:

I have five high'ish spec laptops with Debian, four of which run Debian 10. Two of them got a bad configuration file: I had to erase the .conf file on one and today I replaced the .conf file on another to one that works.

People come to this site because they want to connect a dock. If they could avoid it, they do not particularly want to read all the info I read before I asked for help. I personally don't mind reading a lot of technical stuff as I am used to it, being a long retired (I am 73) Unix systems manager, systems programmer and university lecturer. 
However, I believe that "normal" people where possible much prefer some nice SHORT answers that are relevant for specific computers, so I posted a short comment about a specific computer.

...  "stating why something needs to be done": I did: If you have a Surface Pro 3 with Debian 10 and install displaylink, your dock will NOT connect to your external screen.

...  "what should be done":  I did: Erase the file and it will work.

No offence: I wasn't aware that you intended to change the scripts, since you haven't fixed the installer patch-file issue (I needed that file in all the four laptops I have used displaylink-debian on so far). I am not complaining and it doesn't matter to me; I have the file.

I didn't send a debug file as I had solved the problem. I did list the contents of the bad configurationfile. If you would like a debug file, let me know and I will send you one. I actually did not erase the file, I just commented the lines out.

But let us stop beating the dead horse. All of us appreciate your efforts as you from a technically viewpoint are doing a very, very good job.