AdnanHodzic/displaylink-debian #472: Monitors not detected

padackar, 2020-10-11:
Hello,

building up my new workstation I have found your project, so firstly thank you for your work. 
I have dock model D3100, Debian 10, Lenovo E530 and I have problem with monitors - they are not detected in LXRandr (nor arandr). However, looks like all other ports work on the dock, I can plug there USB-device, audio and they works... 

> $ cat /etc/X11/xorg.conf.d/20-displaylink.conf 
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

> $ sudo systemctl status displaylink-driver.service 
● displaylink-driver.service - DisplayLink Driver Service
   Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vend
   Active: active (running) since Sun 2020-10-11 10:15:05 CEST; 4min 31s ago
  Process: 4122 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install $(ls -t /
 Main PID: 4124 (DisplayLinkMana)
    Tasks: 39 (limit: 4915)
   Memory: 25.8M
   CGroup: /system.slice/displaylink-driver.service
           └─4124 /opt/displaylink/DisplayLinkManager
Oct 11 10:15:05 dev systemd[1]: Starting DisplayLink Driver Service...
Oct 11 10:15:05 dev systemd[1]: Started DisplayLink Driver Service.

> $ sudo ./displaylink-debian.sh --debug
--.------------------------------------------------------
Starting Debug ...
Did you read Post Installation Guide? http://bit.ly/2TbZleK [y/N] y
Did you read Troubleshooting most common issues? http://bit.ly/2Rofd0x [y/N] y
--------------- Linux system info ----------------
Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64
---------------- DisplayLink info ----------------
Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0
------------------ Graphics card -----------------
Vendor: i915
nouveau
Subsystem: Core
GT
VGA: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
NVIDIA Corporation GF108M [GeForce GT 630M] (rev a1)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf
-------------- DisplayLink xorg.conf -------------
File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
-------------------- Monitors --------------------
Providers: number : 6
Provider 0: id: 0x14e cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x127 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 2: id: 0x105 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0xe3 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0xb0 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x42 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 2 outputs: 1 associated providers: 0 name:modesetting

Could you please help me / tell me what I am doing wrong here..?

jdybdahl, 2020-10-11:
You have the providers necessary. Try kngharv's excellent advice https://github.com/AdnanHodzic/displaylink-debian/issues/469

padackar, 2020-10-12:
Works, thank you very much.

closed 2020-10-12 by padackar