AdnanHodzic/displaylink-debian #473: Ryzen 7 laptop AMD Radeon

Barabazs, 2020-10-13:
I'm actually having a bit of the same issue... I can get my Ryzen 7 laptop to do HDMI out separately to USB 3.0 DisplayLink... (Audio is a bitch on each restart... takes DisplayLink over HDMI)... 

Is there a way to Nvidia & DisplayLink to play nicely?

Here are my efforts from the above (but swear it's trying to Radeon Vega me instead of NVidia me...)

xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x27a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x258 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x236 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x214 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f2 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 1 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 2 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 3 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 4 0
robsta@robstas-um433iq:~$ xrandr --setprovideroutputsource 5 0
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  140 (RANDR)
  Minor opcode of failed request:  35 (RRSetProviderOutputSource)
  Value in failed request:  0x1f2
  Serial number of failed request:  20
  Current serial number in output stream:  21
robsta@robstas-um433iq:~$ xrandr --listproviders
Providers: number : 6
Provider 0: id: 0x55 cap: 0x9, Source Output, Sink Offload crtcs: 4 outputs: 2 associated providers: 4 name:Unknown AMD Radeon GPU @ pci:0000:04:00.0
Provider 1: id: 0x27a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 2: id: 0x258 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x236 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 4: id: 0x214 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 5: id: 0x1f2 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G0

_Originally posted by @robotard in https://github.com/AdnanHodzic/displaylink-debian/issue_comments/707480291_

Barabazs, 2020-10-13:
Robotard, DisplayLink says re Ubuntu: "Closed source AMD/NVIDIA drivers are incompatible with DisplayLink driver. Please use open-source AMD/NVIDIA drivers instead."
https://support.displaylink.com/knowledgebase/articles/641668-known-issues-with-displaylink-ubuntu-support

_Originally posted by @jdybdahl in https://github.com/AdnanHodzic/displaylink-debian/issue_comments/707566535_

Barabazs, 2020-10-13:
That’s strange. Working fine on Kernel 4.8. 

_Originally posted by @robotard in https://github.com/AdnanHodzic/displaylink-debian/issue_comments/707741496_

Barabazs, 2020-10-13:
@robotard Don't forget to post the debug output please.

robotard, 2020-10-14:
Assuming you mean from Kernel 5.9?

Or did you want the one working from 5.8.14 with Ryzen Vega 7 (Renoir) + Nvidia MX350 working fine?

robotard, 2020-10-14:
Working Kernel 5.8.14 with above graphics...
[DLSupportTool_Output_2020-10-14T04:14:14.680212.zip](https://github.com/AdnanHodzic/displaylink-debian/files/5375381/DLSupportTool_Output_2020-10-14T04.14.14.680212.zip)

robotard, 2020-10-14:
Well, there's both...

[Kernel-5.8.14-DLSupportTool_Output_2020-10-14T04:14:14.680212.zip](https://github.com/AdnanHodzic/displaylink-debian/files/5375396/Kernel-5.8.14-DLSupportTool_Output_2020-10-14T04.14.14.680212.zip)
[Kernel-5.9-DLSupportTool_Output_2020-10-14T04:18:04.156934.zip](https://github.com/AdnanHodzic/displaylink-debian/files/5375397/Kernel-5.9-DLSupportTool_Output_2020-10-14T04.18.04.156934.zip)

robotard, 2020-10-14:
I received a link to a patch which I'll test later too... Comes with this caveat:

_"If you wish I have a testing patch, untested since I don't have any HW to test right now and will only work with a 5.9 kernel.

Patch against latest release:

https://crazy.dev.frugalware.org/evdi-kernel-5.9-test.patch"_

closed 2020-10-16 by Barabazs