AdnanHodzic/displaylink-debian #474: DisplayLink (latest official) not working with Kernel 5.9

robotard, 2020-10-17:
Hey guys,

I think I opened this before, but actually think that it's all the more relevant now as there appears to be a more recent patch available that could perhaps be included or improved upon? Perhaps if our great minds head over to the discussion we can figure a quicker fix?

It's being discussed (rather slowly) here - https://github.com/DisplayLink/evdi/issues/225

And the discussed patch written by (https://github.com/abucodonosor) is found here - https://crazy.dev.frugalware.org/evdi-kernel-5.9-test.patch

Now, my thoughts, for those more knowledgeable than I, would be to modify the above patch to only perform the relevant on 5.9, and leave the existing on < 5.9 where they work as they currently do... 

Anyone fancy giving that a go? I'm cross linking to Adnan's github in the other discussion also...

Barabazs, 2020-10-17:
I don't understand what you're trying to achieve. What do you want displaylink-debian to do and why?

bnavigator, 2020-10-17:
@robotard

https://github.com/DisplayLink/evdi/pull/228 already does what you are asking for. If you need it as patch file use the URL https://github.com/DisplayLink/evdi/pull/228.patch

robotard, 2020-10-17:
> DisplayLink/evdi#228 already does what you are asking for. If you need it as patch file use the URL https://github.com/DisplayLink/evdi/pull/228.patch

Bring another patch together to get Kernel 5.9 working that doesn’t interrupt older kernels. And works. Pretty much just that. Kinda a community effort...

robotard, 2020-10-17:
Sorry, didn’t see a patch file there. Just discussions about ArchLinux

bnavigator, 2020-10-17:
> Bring another patch together to get Kernel 5.9 working that doesn’t interrupt older kernels. And works. Pretty much just that. Kinda a community effort...

The patch doesn't interrupt older kernels. It just ADDS support for kernel 5.9

bnavigator, 2020-10-17:
> Sorry, didn’t see a patch file there. Just discussions about ArchLinux

Every PR is a patch

robotard, 2020-10-19:
Has anyone else had success with this patch, Ubuntu & 5.9? Had a few
hiccups along the way (my own typo), but applied and replaced evdi, but
still causing me a little bit of a headache.

Have attached my logs.

Barabazs, 2020-10-19:
> Has anyone else had success with this patch, Ubuntu & 5.9? Had a few hiccups along the way (my own typo), but applied and replaced evdi, but still causing me a little bit of a headache. Have attached my logs.

There are no logs in your post?

There are already several persons helping you in the issue where the patch originated. You should continue the discussion there and only there.

closed 2020-10-19 by Barabazs