AdnanHodzic/displaylink-debian #476: Running Debian Buster with two external displays not being detected.

msd90, 2020-10-26:
Currently running a Dell Inspirion 14 Laptop connected to two external displays. External displays are not detected.

```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xrdp/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "Device"
    Identifier  "Intel"
    Driver      "intel"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

Barabazs, 2020-10-26:
Replace your 20-displaylink.conf with:
```
Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection
```

msd90, 2020-10-26:
> Replace your 20-displaylink.conf with:
> 
> ```
> Section "OutputClass"
>     Identifier  "DisplayLink"
>     MatchDriver "evdi"
>     Driver      "modesetting"
>     Option      "AccelMethod" "none"
> EndSection
> ```

Still no change.

Barabazs, 2020-10-26:
Repost the debug output please.

msd90, 2020-10-26:
```
-------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xrdp/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf /etc/X11/xorg.conf.d/00-keyboard.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------
```

Barabazs, 2020-10-27:
Can you post the content of the other configs please?
/etc/X11/xrdp/xorg.conf
/etc/X11/xorg.conf.d/00-keyboard.conf

msd90, 2020-10-27:
/etc/X11/xrdp/xorg.conf
```
Section "ServerLayout"
    Identifier "X11 Server"
    Screen "Screen (xrdpdev)"
    InputDevice "xrdpMouse" "CorePointer"
    InputDevice "xrdpKeyboard" "CoreKeyboard"
EndSection

Section "ServerFlags"
    Option "DontVTSwitch" "on"
    Option "AutoAddDevices" "off"
EndSection

Section "Module"
    Load "dbe"
    Load "ddc"
    Load "extmod"
    Load "glx"
    Load "int10"
    Load "record"
    Load "vbe"
    Load "xorgxrdp"
    Load "fb"
EndSection

Section "InputDevice"
    Identifier "xrdpKeyboard"
    Driver "xrdpkeyb"
EndSection

Section "InputDevice"
    Identifier "xrdpMouse"
    Driver "xrdpmouse"
EndSection

Section "Monitor"
    Identifier "Monitor"
    Option "DPMS"
    HorizSync 30-80
    VertRefresh 60-75
    ModeLine "1920x1080" 138.500 1920 1968 2000 2080 1080 1083 1088 1111 +hsync -vsync
    ModeLine "1280x720" 74.25 1280 1720 1760 1980 720 725 730 750 +HSync +VSync
    Modeline "1368x768" 72.25 1368 1416 1448 1528 768 771 781 790 +hsync -vsync
    Modeline "1600x900" 119.00 1600 1696 1864 2128 900 901 904 932 -hsync +vsync
EndSection

Section "Device"
    Identifier "Video Card (xrdpdev)"
    Driver "xrdpdev"
EndSection

Section "Screen"
    Identifier "Screen (xrdpdev)"
    Device "Video Card (xrdpdev)"
    Monitor "Monitor"
    DefaultDepth 24
    SubSection "Display"
        Depth 24
        Modes "640x480" "800x600" "1024x768" "1280x720" "1280x1024" "1600x900" "1920x1080"
    EndSubSection
EndSection
```
 /etc/X11/xorg.conf.d/00-keyboard.conf
```
# Read and parsed by systemd-localed. It's probably wise not to edit this file
# manually too freely.
Section "InputClass"
        Identifier "system-keyboard"
        MatchIsKeyboard "on"
        Option "XkbLayout" "us"
        Option "XkbModel" "pc105"
EndSection
```

Barabazs, 2020-10-27:
Do you need those files for your peripherals?

I would try to exclude those files (append .backup or whatever to the filename). 
Reboot and see if the debug output changes. If yes, repost it please.

msd90, 2020-10-27:
No change still the same:

```
--------------- Linux system info ----------------

Distro: Debian
Release: buster
Kernel: 4.19.0-11-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: Graphics
VGA: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
VGA (3D): 
X11 version: 1.20.4-1+deb10u1
X11 configs: /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "OutputClass"
    Identifier  "DisplayLink"
    MatchDriver "evdi"
    Driver      "modesetting"
    Option      "AccelMethod" "none"
EndSection

-------------------- Monitors --------------------

Providers: number : 1
Provider 0: id: 0x46 cap: 0xb, Source Output, Sink Output, Sink Offload crtcs: 4 outputs: 3 associated providers: 0 name:Intel

-------------------------------------------------------------------

```

Barabazs, 2020-10-28:
Maybe retry with the original config, reboot, test, etc

msd90, 2020-10-29:
Nothing changed. I'm just using my HDMI cable to connect to one of the external monitors at the moment.

closed 2021-01-17 by Barabazs