AdnanHodzic/displaylink-debian #477: Asus mb169+ as plus monitor on parrot os

KerZsoo, 2020-10-26:
Helloo.
Can anyone help me pls?
I got an Asus  mb169b+ portable monitor.
I have an Asus rog gl503ge laptop with windows 10 on it, and i have an external ssd with fully installed parrot os.
So on windows 10 the mb169+ work good, but when i boot parrot it do not recognize the monitor only  in "lsusb".
here is some otuput:
(sorry for bad english :( )

$ uname -a
Linux parrot 5.8.0-2parrot1-amd64 #1 SMP Debian 5.8.10-2parrot1 (2020-10-05) x86_64 GNU/Linux

$ xrandr --listproviders 
Providers: number : 6
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x317 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G4
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

$ lsusb 
Bus 002 Device 002: ID 0b05:1932 ASUSTek Computer, Inc. ROG STRIX Arion
Bus 002 Device 003: ID 17e9:ff0b DisplayLink ASUSTEK MB169B+
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0b05:1869 ASUSTek Computer, Inc. ITE Device(8910)
Bus 001 Device 003: ID 13d3:5666 IMC Networks USB2.0 HD UVC WebCam
Bus 001 Device 005: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
Bus 001 Device 002: ID 1532:0039 Razer USA, Ltd Orochi 2013
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ ./displaylink-debian.sh --debug 
-------------------------------------------------------------------

Starting Debug ...

--------------- Linux system info ----------------

Distro: Parrot
Release: n/a
Kernel: 5.8.0-2parrot1-amd64

---------------- DisplayLink info ----------------

Driver version: 1.7.0
DisplayLink service status: up and running
EVDI service version: 1.7.0

------------------ Graphics card -----------------

Vendor: i915
Subsystem: UHD
VGA: Intel Corporation UHD Graphics 630 (Mobile)
VGA (3D): NVIDIA Corporation GP107M [GeForce GTX 1050 Ti Mobile] (rev a1)
X11 version: 1.20.8-2
X11 configs: /etc/X11/xorg.conf.d/xorg.conf /etc/X11/xorg.conf.d/20-displaylink.conf

-------------- DisplayLink xorg.conf -------------

File: /etc/X11/xorg.conf.d/20-displaylink.conf
Contents:
 Section "ServerLayout"
    Identifier "layout"
    Screen 0 "intel"
    Screen 1 "nvidia
    #Inactive "intel"
EndSection

Section "Device"
    Identifier "intel"
    Driver "modesetting"
   #Option "AccelMethod" "None"
   #    hozzzaadott
   # Driver "intel"
   # BusID   "PCI:0:2:0"
   # Option "AccelMethod"    "uxa"
EndSection

Section "Screen"
    Identifier "intel"
    Device "intel"
EndSection

Section "Device"
    Identifier "nvidia"
    Driver "nvidia"
    Option "ConstrainCursor" "off"
EndSection

Section "Screen"
    Identifier "nvidia"
    Device "nvidia"
    Option "AllowEmptyInitialConfiguration" "on"
    Option "IgnoreDisplayDevices" "CRT"
EndSection

-------------------- Monitors --------------------

Providers: number : 6
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 0 name:modesetting
Provider 1: id: 0x317 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G4
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 3: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------
please, i realy need help,unfortunately i am a complete noob to linux.
i think my  20-displaylink.conf is a mess, when i isntelled the displaylink-debian.sh my laptop screen turn black, so i commented that line : ( #Inactive "intel" )...

Barabazs, 2020-10-27:
Did you perform the post install steps? [post-install-guide.md#screen-layout-configuration
](https://github.com/AdnanHodzic/displaylink-debian/blob/master/post-install-guide.md#screen-layout-configuration)

KerZsoo, 2020-10-27:
Ok, now i did the:

$ xrandr --setprovideroutputsource 1 0                                                                                                                                                                     
X Error of failed request:  BadValue (integer parameter out of range for operation)                                                                                                                            Major opcode of failed request:  140 (RANDR)                                                                                                                                                                 
Minor opcode of failed request:  35 (RRSetProviderOutputSource)                                                                                                                                              
Value in failed request:  0x317                                                                                                                                                                              
Serial number of failed request:  20                                                                                                                                                                         
Current serial number in output stream:  21

then:

$ xrandr --setprovideroutputsource 2 0
(with no output in terminel, i think it is good )

then i did:

$ xrandr --listproviders 
Providers: number : 6
Provider 0: id: 0x46 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 3 outputs: 4 associated providers: 1 name:modesetting
Provider 1: id: 0x317 cap: 0x0 crtcs: 0 outputs: 0 associated providers: 0 name:NVIDIA-G4
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

(here is something new : "Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting" compare to the original post.)

xrandr --listmonitors 
Monitors: 1
 0: +*eDP-1 1920/344x1080/193+0+0  eDP-1
(monitor plugged in, when i plug it white leg flash, ASUS logon flash, then turn black and the led goes to orange color..)

$lsusb 
Bus 002 Device 004: ID 17e9:ff0b DisplayLink ASUSTEK MB169B+

$ ./displaylink-debian.sh --debug 
--------------- Linux system info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Distro: Parrot                                                                                                                                                                                                 
Release: n/a                                                                                                                                                                                                   
Kernel: 5.8.0-2parrot1-amd64                                                                                                                                                                                   
                                                                                                                                                                                                               
---------------- DisplayLink info ----------------                                                                                                                                                             
                                                                                                                                                                                                               
Driver version: 1.7.0                                                                                                                                                                                          
DisplayLink service status: up and running                                                                                                                                                                     
EVDI service version: 1.7.0                                                                                                                                                                                    
                                                                                                                                                                                                               
------------------ Graphics card -----------------                                                                                                                                                             
                                                                                                                                                                                                               
[... cut ...]
Provider 2: id: 0x17e cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 1 name:modesetting
Provider 3: id: 0x15c cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 4: id: 0x13a cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
Provider 5: id: 0x118 cap: 0x2, Sink Output crtcs: 1 outputs: 1 associated providers: 0 name:modesetting

-------------------------------------------------------------------

so there is some progress thank you :)
what is the next step?

Barabazs, 2020-10-27:
It's explained in more detail here: https://github.com/AdnanHodzic/displaylink-debian/issues/469

KerZsoo, 2020-10-27:
wow
it is working!!! very very thank you :)
I will figure out the persistence steps for me,
incredibly grateful thanks for the help :)

closed 2020-10-27 by Barabazs