AdnanHodzic/displaylink-debian #480: evdi won't compile on bullseys's new 5.9 kernel

tobias-weiss, 2020-10-27:
I have to stick to 5.8 kernel, there the module does work.

`Klone nach 'evdi' ...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (46/46), done.
remote: Total 46 (delta 2), reused 9 (delta 0), pack-reused 0
Entpacke Objekte: 100% (46/46), 66.98 KiB | 629.00 KiB/s, Fertig.
Hinweis: Wechsle zu 'dc595db636845aef39490496bc075f6bf067106c'.

Sie befinden sich im Zustand eines 'losgelösten HEAD'. Sie können sich
umschauen, experimentelle Änderungen vornehmen und diese committen, und
Sie können alle möglichen Commits, die Sie in diesem Zustand machen,
ohne Auswirkungen auf irgendeinen Branch verwerfen, indem Sie zu einem
anderen Branch wechseln.

Wenn Sie einen neuen Branch erstellen möchten, um Ihre erstellten Commits
zu behalten, können Sie das (jetzt oder später) durch Nutzung von
'switch' mit der Option -c tun. Beispiel:

  git switch -c <neuer-Branchname>

Oder um diese Operation rückgängig zu machen:
  git switch -

Sie können diesen Hinweis ausschalten, indem Sie die Konfigurationsvariable
'advice.detachedHead' auf 'false' setzen.

patching file .travis.yml
Hunk #1 FAILED at 26.
1 out of 1 hunk FAILED -- saving rejects to file .travis.yml.rej
patching file Makefile
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file Makefile.rej
patching file library/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file library/Makefile.rej
patching file module/Makefile
Hunk #1 FAILED at 43.
1 out of 1 hunk FAILED -- saving rejects to file module/Makefile.rej
patching file module/evdi_drv.c
Hunk #1 FAILED at 81.
1 out of 1 hunk FAILED -- saving rejects to file module/evdi_drv.c.rej
patching file module/evdi_drv.h
Hunk #1 FAILED at 20.
Hunk #2 FAILED at 51.
Hunk #3 FAILED at 112.
3 out of 3 hunks FAILED -- saving rejects to file module/evdi_drv.h.rej
patching file module/evdi_gem.c
Hunk #1 FAILED at 61.
Hunk #2 FAILED at 254.
Hunk #3 FAILED at 569.
Hunk #4 FAILED at 581.
4 out of 4 hunks FAILED -- saving rejects to file module/evdi_gem.c.rej
CFLAGS="-Werror -Wextra -Wall -Wmissing-prototypes -Wstrict-prototypes -Wno-error=missing-field-initializers" make -C module 
make[1]: Verzeichnis „/home/tobias/Downloads/displaylink-debian/evdi/module“ wird betreten
make -C /lib/modules/5.9.0-1-amd64/build M=$PWD
make[2]: Verzeichnis „/usr/src/linux-headers-5.9.0-1-amd64“ wird betreten
  AR      /home/tobias/Downloads/displaylink-debian/evdi/module/built-in.a
  CC [M]  /home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.o
/home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:3: error: ‘struct drm_driver’ has no member named ‘gem_free_object’; did you mean ‘gem_open_object’?
   92 |  .gem_free_object = evdi_gem_free_object,
      |   ^~~~~~~~~~~~~~~
      |   gem_open_object
/home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:21: error: initialization of ‘void (*)(struct drm_device *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
   92 |  .gem_free_object = evdi_gem_free_object,
      |                     ^~~~~~~~~~~~~~~~~~~~
/home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:21: note: (near initialization for ‘driver.lastclose’)
/home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.c: In function ‘evdi_platform_probe’:
/home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.c:173:20: error: ‘struct dev_archdata’ has no member named ‘iommu’
  173 |  pdev->dev.archdata.iommu = INTEL_IOMMU_DUMMY_DOMAIN;
      |                    ^
cc1: all warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-5.9.0-1-common/scripts/Makefile.build:288: /home/tobias/Downloads/displaylink-debian/evdi/module/evdi_drv.o] Fehler 1
make[3]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:1796: /home/tobias/Downloads/displaylink-debian/evdi/module] Fehler 2
make[2]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:185: __sub-make] Fehler 2
make[2]: Verzeichnis „/usr/src/linux-headers-5.9.0-1-amd64“ wird verlassen
make[1]: *** [Makefile:71: module] Fehler 2
make[1]: Verzeichnis „/home/tobias/Downloads/displaylink-debian/evdi/module“ wird verlassen
make: *** [Makefile:8: all] Fehler 2
`

Barabazs, 2020-10-27:
evdi are still working on supporting 5.9

mndz, 2020-12-02:
I'm getting the same error

Barabazs, 2020-12-05:
@tobias-weiss @mndz support for 5.9 has been added.
pull the latest from master and give it a try.

closed 2020-12-05 by Barabazs