AdnanHodzic/displaylink-debian #481: Cannot install on Debian, installer script fails

mbrennwa, 2020-10-28:
I followed the install instructions (I am running Debian):
```
git clone https://github.com/AdnanHodzic/displaylink-debian.git
cd displaylink-debian/ && sudo ./displaylink-debian.sh
```

The installer script fails with this:
```/home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:3: error: ‘struct drm_driver’ has no member named ‘gem_free_object’; did you mean ‘gem_open_object’?
   92 |  .gem_free_object = evdi_gem_free_object,
      |   ^~~~~~~~~~~~~~~
      |   gem_open_object
/home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:21: error: initialization of ‘void (*)(struct drm_device *)’ from incompatible pointer type ‘void (*)(struct drm_gem_object *)’ [-Werror=incompatible-pointer-types]
   92 |  .gem_free_object = evdi_gem_free_object,
      |                     ^~~~~~~~~~~~~~~~~~~~
/home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.c:92:21: note: (near initialization for ‘driver.lastclose’)
/home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.c: In function ‘evdi_platform_probe’:
/home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.c:173:20: error: ‘struct dev_archdata’ has no member named ‘iommu’
  173 |  pdev->dev.archdata.iommu = INTEL_IOMMU_DUMMY_DOMAIN;
      |                    ^
cc1: all warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-5.9.0-1-common/scripts/Makefile.build:288: /home/mbrennwa/Downloads/displaylink-debian/evdi/module/evdi_drv.o] Error 1
make[3]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:1796: /home/mbrennwa/Downloads/displaylink-debian/evdi/module] Error 2
make[2]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:185: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.9.0-1-amd64'
make[1]: *** [Makefile:71: module] Error 2
make[1]: Leaving directory '/home/mbrennwa/Downloads/displaylink-debian/evdi/module'
make: *** [Makefile:8: all] Error 2
cp: cannot stat 'library/libevdi.so.1.*': No such file or directory
```

My brand-new i-tec docking station is a useless brick...

Barabazs, 2020-10-28:
evdi are still working on supporting 5.9

mbrennwa, 2020-10-28:
If this software does not work on certain kernel versions, the installer should at least check for the kernel versions and provide a meaningful error message to the user.

Barabazs, 2020-10-28:
You're always welcome to submit a PR.

mbrennwa, 2020-10-28:
What do you mean by "PR"?

closed 2020-10-28 by Barabazs