AdnanHodzic/displaylink-debian #482: Display not recignized, remains dark

mbrennwa, 2020-10-28:
After downgrading the kernel to 5.8 I managed to install the software on Debian. However, the external display is not recognized and remains dark.

```
systemctl status displaylink-driver.service
● displaylink-driver.service - DisplayLink Driver Service
     Loaded: loaded (/lib/systemd/system/displaylink-driver.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2020-10-28 11:20:41 CET; 135ms ago
    Process: 6489 ExecStartPre=/bin/sh -c modprobe evdi || (dkms install $(ls -t /usr/src | grep evdi | head -n1  | sed -e "s:-:/:") && modprobe evdi) (code=exited, status=1/FAILURE)
```

Barabazs, 2020-10-28:
Please post the debug output.

mbrennwa, 2020-10-28:
How / where do I get the debug output?

Barabazs, 2020-10-29:
It's clearly documented in the readme.

mbrennwa, 2020-10-29:
Sorry, too late. I returned the i-tec docking station and replaced it with a different unit that just works out of the box without any third party drivers.

closed 2020-10-29 by Barabazs