AdnanHodzic/displaylink-debian #483: Error while installing EVDI library

Tavs9, 2020-10-28:
Hello,

I'm trying to install the Driver on my NUC6YAB running on Linux Debian 9.11 and Kernel 4.19 but I've got an error on "Installing EVDI library" step. Here is the message I have :

**[[ Installing EVDI library ]]
make: Nothing to be done for 'default'.
cp: cannot stat 'libevdi.so': No such file or directory
ERROR: Failed to copy evdi/5.3.1.34 library to /opt/displaylink.**

I found the "livevdo.so" file I tried to copy it in different folders but it doesn't change anything.

Does someone has a solution for this problem ?

Here is the whole installation details in attachment if necessessary :
[Displaylink-debian installation2.txt](https://github.com/AdnanHodzic/displaylink-debian/files/5453027/Displaylink-debian.installation2.txt)
 

Thanks

Barabazs, 2020-10-29:
Why are you running this?
`root@debian:/opt# wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink-debian.sh && wget https://raw.githubusercontent.com/AdnanHodzic/displaylink-debian/master/displaylink.sh && sed -i 's/\.\/evdi.sh/\#\.\/evdi.sh/g' displaylink-debian.sh && wget https://github.com/DisplayLink/evdi/archive/v1.6.4.tar.gz && tar -zxvf v1.6.4.tar.gz && mv evdi-1.6.4/ evdi && cd evdi/ && wget https://crazy.dev.frugalware.org/evdi-all-in-one-fixes.patch && nano .travis.yml
`

Tavs9, 2020-10-29:
I run this because I have the problem concerning the "EVDI patch failure" explained in the issue #398
This steps are given in this issue to resolve the problem.

Barabazs, 2020-10-29:
That has been resolved since then. 

What did the "regular" install output when you ran it?

Tavs9, 2020-10-29:
Okay, I don't really understand why but it worked. Perhaps the fact I've updated the Kernel for other issues after the first tries has solved the problem.
Sorry for this avoidable issue and thanks for your reactivity

closed 2020-10-29 by Barabazs